ros-planning/navigation #302: param to prevent updateCosts() in layer

aswinthomas, 2015-01-26:
Currently the only way to retain information in a costmap layer, and not copy it into the master layer is to write a custom plugin with no updateCosts() method implemented. 

Could we have a parameter that prevents updateWithOverwrite() or updateWithMax()?

DLu, 2015-01-30:
Is the idea that you can dynamically switch between layer 1 and layer 2? Otherwise I'm not sure why you would store things in a layer in the first place.

aswinthomas, 2015-01-31:
Lets say we have 5 layers having obstacles that you should not cross e.g. walls
and 2 layers with zones that should be avoided but not impossible to cross e.g. lane markings

In the current version, all the information in these layers are copied over to the master map. making the robot avoid obstacles in all the above layers. 
If we have a param called copy_to_master = false for the last 2 layers, then the planner will avoid only the first 5 layers. In addition to this, the planner can extract data from the last 2 layers and perform additional behaviors/costs for them.

DLu, 2015-02-10:
Would something like this work?
https://github.com/DLu/navigation/compare/no_combine

aswinthomas, 2015-02-11:
I think so, yes!

mikeferguson, 2015-02-16:
Addressed by #306

closed 2015-02-16 by mikeferguson