ros-planning/navigation #308: Compilation error with openni

edran, 2015-02-16:
When I try to my code (which uses costmap_2d) I get this error:

```
CMake Error at /opt/ros/hydro/share/costmap_2d/cmake/costmap_2dConfig.cmake:106 (message):
  Project 'costmap_2d' specifies '/usr/include/ni' as an include dir, which
  is not found.  It does neither exist as an absolute directory nor in
  '/opt/ros/hydro//usr/include/ni'.  Ask the maintainer 'David V.  Lu!!
  <davidvlu@gmail.com>, Michael Ferguson <fergs@unboundedrobotics.com>' to
  fix it.
Call Stack (most recent call first):
  /opt/ros/hydro/share/catkin/cmake/catkinConfig.cmake:75 (find_package)
  test_map/CMakeLists.txt:7 (find_package)
```
I'm on precise/hydro, with latest packages (at least from the Ubuntu ROS PPA).

Compiling OpenNi from source and installing it **fixed** the issue.

mikeferguson, 2015-02-16:
I presume this gets dragged in from "pcl_ros" -- our CMake does not specify this directory (which also means we can't fix it -- and this is building on the ROS farm, so I don't think it is a missing dependency).

I'm guessing you didn't have all (build) dependencies installed. You can run "rosdep install --from-paths src --ignore-src --rosdistro hydro -y" in your workspace to install build-time dependencies required for a source build (I presume if you run that now, you'll see something like libopenni2-dev or another -dev package installed).

closed 2015-02-16 by mikeferguson

edran, 2015-02-19:
Mmmh, weird - I thought I had roughly run the same command. I'll check again with a clean installation just in case.
Thanks though.