ros-planning/navigation #313: robot_pose_ekf: PoseWithCovarianceStamped message hard coded to odom

hsmrs, 2015-02-23:
Hey everyone,

I noticed that in line 393 of odom_estimation.cpp in robot_pose_ekf the PoseWithCovarianceStamped message is hard coded to the "odom" frame instead of using the output_frame_ variable. This seems to me like a bug. I am currently working around it by creating a node to publish the message with the correct frame, but it would be nice to have this fixed.

Thanks!

mikeferguson, 2015-02-26:
It seems that maybe it should be set to "output_frame_"? I don't have any time to debug and be sure -- but tested pull requests are definitely very welcome.

hsmrs, 2015-02-27:
That is my impression as well. Unfortunately I am also in your situation, with no time to debug and test properly if I made the change.

v01d, 2015-08-03:
Found the same issue. I think it is a trivial change, this is just output data, so I think it couldn't really affect anything.

emilyfy, 2018-01-27:
I found it too after wondering why the published topic is on "odom" frame. Tried changing it to output_frame_ and it fixes it and seems to work well.

commit 8146332a (2018-02-13): Fixes #313 (indigo) (#653)
commit 42d4cca2 (2018-02-13): Fixes #313 (kinetic) (#654)
commit c1b90ce5 (2018-02-13): Fixes #313 (Lunar) (#655)
closed 2018-02-13 by DLu