ros-planning/navigation #327: ObstacleCostFunction has an empty footprint when the first goal is an in-place rotation

ex-ratt, 2015-04-02:
There seems to be a problem with setting the footprint to the ObstacleCostFunction when using dwa_local_planner. The footprint is set whenever DWAPlanner::findBestPath is called (line 298 in https://github.com/ros-planning/navigation/blob/indigo-devel/dwa_local_planner/src/dwa_planner.cpp) and nowhere else (to the best of my knowledge). In case the goal given to the robot is at its position, so it only has to rotate to match the orientation, DWAPlanner::findBestPath is not called - instead, the LatchedStopRotateController only calls DWAPlanner::checkTrajectory to check for a valid rotation command, which does not set the footprint to the ObstacleCostFunction, though it needs the footprint to check for valid rotation.

If now the very first goal to the robot after starting the move_base node results in such an in-place rotation, then the ObstacleCostFunction cannot check the rotation command and instead prints an error to the log / console (see line 78ff in https://github.com/ros-planning/navigation/blob/indigo-devel/base_local_planner/src/obstacle_cost_function.cpp). If another command was given beforehand, where the robot had to move, then this error does not appear, as the footprint was set. This problem occurred in a question: http://answers.ros.org/question/204740/dwa_local_planner-in-place-rotation-first-goal-does-not-work/?answer=205955#post-id-205955

I wonder if it were possible to set the footprint at construction time. The DWAPlanner knows the costmap and calls the ObstacleCostFunction's constructor with that costmap and the costmap should know the footprint (I'm just not sure if it knows at that moment). So it may be possible to retrieve and set the footprint in the constructor of ObstacleCostFunction. If this is not possible, then it has to be set in DWAPlanner::checkTrajectory similar to DWAPlanner::findBestPath, but that would mean that quite some other objects have to now about the footprint, which to me does not seem that desirable.

Swan-Baigne, 2015-05-19:
Hello,
i have the same problem, and i wanted to know if any modification was planned, or if i have to rebuild the dwa_planner package myself

ZdenekM, 2015-08-13:
I'm also getting "Footprint spec is empty, maybe missing call to setFootprint?" right now so I vote +1 for this issue...

askkl, 2015-10-27:
I have the same problem, too

bradleypowers, 2016-01-11:
@DLu Any plans to fix this?  We're seeing the same issue as well.  We may be able to take a look at a fix, but bandwidth is limited here atm.

Swan-Baigne, 2016-01-12:
From the time of the first post, it seems it won't be fixed anytime soon. I work around the problem by forcing the robot to move forward a little in the initialisation, so that the footprint is created.

mikeferguson, 2016-01-12:
I'm adding a new "help wanted" tag to this ticket -- as neither of the maintainers currently have the cycles to work on it. A tested pull request would be greatly appreciated.

corot, 2016-04-05:
@ex-ratt is right, so  I just do what he says, set the footprint on dynamic reconfigure, and it works for me. However, this is a simplistic fix, without deep study of the problem, so please give it a try before merging!

pgigioli, 2016-06-16:
@corot, I tried your fix and it is working perfectly so far.  Good job.

hvpandya, 2016-06-22:
While the fix suggested by @corot does resolve this specific issue, I would rather not have footprint_spec passed as an argument to DWAPlanner::reconfigure, it just does not feel right. 

I wonder if it can be a part of  base_local_planner::LocalPlannerUtil which is already passed to the DWAPlanner constructor. Or may be have a separate PlannerUtil for dwa which uses costmap_2d::Costmap2DROS instead of costmap_2d::Costmap2D. Any negative impacts of implementing the latter? I might give it a try when I get some spare time.

corot, 2016-06-26:
Hi @hvpandya, Yes, fully agree that my PR is not the most elegant change. I just wanted to fix the problem without changing significantly the code. I didn't try your fix, but if it solves the issue, feel free to PR it and I'll drop my own, as your change is probably cleaner

MarcelSoler, 2019-01-02:
Hi! Any updates? is #681 ready to merge? I would like to use it. If some help is needed I could lend you a hand to speed up the process. 
Just let me know! thanks.

MarcelSoler, 2019-01-03:
If it's ok I'll fork it, test it and re-summit a pull request.

MarcelSoler, 2019-01-08:
Opened new pull request continuing #681 #829 taking feedback into account. Tested ok!

DLu, 2019-02-11:
Fixes in #829 #859 #860 #861

closed 2019-02-11 by DLu