ros-planning/navigation #328: Replanning Issue

EMolinos, 2015-04-07:
Hello:

I'm trying to configure move_base with my mobile robot platform and i am facing a serious issue. I am using ROS Hydro and navigation stack 1.11.15

I have configured the navigation package to follow the generated path as close as possible and when an unexpected obstacle appears in the path from outside the local window, the navfn and global_path is replanned and everything works OK. But if the obstacle appear near the robot (where the global path is already planned) it is not able to replan and approaches the obstacle until the robot stop.

If I configure the global planner to replan at a certain ratio, it is able to replan avoiding the obstacle, but this is not the behaviour (periodic planning) we want in the real application, only replanning when the path is blocked.

My configuration files are as following:

Base local planner params:

base_global_planner: navfn/NavfnROS
base_local_planner: base_local_planner/TrajectoryPlannerROS
recovery_behaviors:  [{name: conservative_reset, type: clear_costmap_recovery/ClearCostmapRecovery}, {name: rotate_recovery, type: rotate_recovery/RotateRecovery}, {name: aggressive_reset, type: clear_costmap_recovery/ClearCostmapRecovery}]
controller_frequency: 10.0
planner_patience: 3.0
controller_patience: 5.0
conservative_reset_dist: 3.0
recovery_behavior_enabled: true
clearing_rotation_allowed: true
shutdown_costmaps: false
oscillation_timeout: 0.0
oscillation_distance: 0.5
planner_frequency: 0.0
global_frame_id: map_navigation

TrajectoryPlannerROS:

  acc_lim_x: 0.4
  acc_lim_y: 0.4
  acc_lim_theta: 0.8

  max_vel_x: 0.2
  min_vel_x: 0.1
  max_trans_vel: 0.2
  min_trans_vel: 0.1 

  max_rotational_vel: 0.6
  max_vel_theta: 0.6
  min_vel_theta: -0.6
  min_in_place_vel_theta: 0.3
  escape_vel: 0.0

  holonomic_robot: false
  y_vels: []
  xy_goal_tolerance: 0.5
  yaw_goal_tolerance: 0.3
  latch_xy_goal_tolerance: true

  sim_time: 1.0
  sim_granularity: 0.025
  angular_sim_granularity: 0.025
  vx_samples: 3
  vtheta_samples: 20
  controller_frequency: 10

  public_cost_grid_pc: true
  meter_scoring: true

  #DWA

  heading_scoring: false
  dwa: false
  forward_point_distance: 0.325

  path_distance_bias: 32
  goal_distance_bias: 24

  #TRAJECTORY PLANNER

  pdist_scale: 0.9
  gdist_scale:  0.8
  occdist_scale: 0.01

  heading_lookahead: 0.325

  publish_cost_grid_pc: false
  global_frame_id: map_navigation

  oscillation_reset_dist: 0.05
  prune_plan: true

NavfnROS:
  allow_unknown: false
  planner_window_x: 0.0
  planner_window_y: 0.0
  default_tolerance: 0.0
  visualize_potential: false

Local costmap params:

local_costmap:
  # Coordinate frame and TF parameters
  global_frame: map_navigation
  robot_base_frame: base_link
  transform_tolerance: 1.0
  # Rate parameters
  update_frequency: 2.0
  publish_frequency: 2.0

  #Map management parameters
  rolling_window: true
  width: 8.0
  height: 8.0
  resolution: 0.05
  origin_x: 0.0
  origin_y: 0.0
  static_map: false

  obstacle_layer:
    observation_sources: laser_scan_sensor
    laser_scan_sensor: {sensor_frame: hokuyo_link, data_type: LaserScan, topic: hokuyo/scan, marking: true, clearing: true, observation_persistence: 0.0, expected_update_rate: 0.0, max_obstacle_height: 2.0, min_obstacle_height: -2.0, obstacle_range: 4.0, raytrace_range: 5.0, inf_is_valid: false}
    max_obstacle_height: 2.0
    obstacle_range: 4.0
    raytrace_range: 5.0
    track_unknown_space: false

  inflation_layer:
    inflation_radius: 5.52
    cost_scaling_factor: 2.0

  plugins:
   - 
     name: obstacle_layer
     type: "costmap_2d::ObstacleLayer"
   - 
     name: inflation_layer
     type: "costmap_2d::InflationLayer"

Global costmap params:

global_costmap:
  # Coordinate frame and TF parameters
  global_frame: map_navigation
  robot_base_frame: base_link
  transform_tolerance: 1.0
  # Rate parameters
  update_frequency: 5.0
  publish_frequency: 2.0

  #Map management parameters
  rolling_window: false
  resolution: 0.05
  static_map: true

  #Static Layer
  static_layer:
    unknown_cost_value: -1
    lethal_cost_threshold: 100
    map_topic: map_navigation

  obstacle_layer:
    observation_sources: laser_scan_sensor
    laser_scan_sensor: {sensor_frame: hokuyo_link, data_type: LaserScan, topic: hokuyo/scan, marking: true, clearing: true, observation_persistence: 0.0, expected_update_rate: 0.0, max_obstacle_height: 2.0, min_obstacle_height: -2.0, obstacle_range: 4.0, raytrace_range: 5.0, inf_is_valid: false}
    max_obstacle_height: 2.0
    obstacle_range: 4.0
    raytrace_range: 5.0
    track_unknown_space: false

  inflation_layer:
    inflation_radius: 5.52
    cost_scaling_factor: 2.0

  plugins:
   - name: static_layer
     type: "costmap_2d::StaticLayer"
   - 
     name: obstacle_layer
     type: "costmap_2d::ObstacleLayer"
   - 
     name: inflation_layer
     type: "costmap_2d::InflationLayer"

Thank you in advance for any idea about how to solve it.

makokal, 2015-09-10:
If your are still having issues with re-planning, just modify your `planner_frequency: 0.0` to a value higher than `0`.

mikeferguson, 2016-02-13:
not really actionable, probably more of a ROS answers question

closed 2016-02-13 by mikeferguson