ros-planning/navigation #333: Publishing Rolling Costmaps Incrementally

DLu, 2015-04-29:
[As seen here](http://answers.ros.org/question/207755/rviz-doesnt-update-rolling-window-costmap-origin/)

Using [OccupancyGridUpdate](https://github.com/ethz-asl/map_msgs/blob/master/msg/OccupancyGridUpdate.msg) will publish the data of the costmap incrementally, but there is currently no way using that message to update the [OccupancyGrid's origin pose](http://docs.ros.org/api/nav_msgs/html/msg/MapMetaData.html)

That data could also be published as a Pose message, but that makes a three-topic channel from what used to be one (bloated) channel.

DLu, 2015-04-29:
@stephanemagnenat Thoughts?

stephanemagnenat, 2015-05-04:
Updating the pose would imply shifting all data. What use case do you envision?

closed 2015-07-16 by mikeferguson