ros-planning/navigation #337: Warning message prints at high frequency, cluttering logging

kogut, 2015-05-07:
If a goal is sent off the global costmap, a warning is printed at very high frequency.   This warning should be throttled to a reasonable rate so other log messages are visible. 

E.g.: 

[ WARN] [1431032125.906108705]: The goal sent to the navfn planner is off the global costmap. Planning will always fail to this goal.
[ WARN] [1431032125.908710995]: The goal sent to the navfn planner is off the global costmap. Planning will always fail to this goal.

DLu, 2015-05-10:
I believe that is shown once per attempted plan. What is your planner_frequency?

kogut, 2015-05-11:
David,

I implemented the "2D Navigation stack" according to:  http://wiki.ros.org/navigation/Tutorials/RobotSetup

When using the RViz "2D Nav Goal" button, if a goal point is specified "off the map" the warning will output at several hundred Hz continuously until a valid goal point is specified.   

Admittedly I have not plunged fully into the details of the tutorial configuration yet to determine if the behavior is really the responsible of something upstream of navfn, e.g. a failed plan continually being resubmitted.

closed 2015-05-19 by DLu