ros-planning/navigation #361: local_planner's control loop is executed very slow on PR2 hydro

furushchev, 2015-06-18:
I'm using PR2 robot on ROS Hydro.
On passing navigation goal, it fails to meet the control loop speed of 10Hz (on my PR2 env, it takes about 1.3 sec on each local control loop), resulting in choppy motion and poor navigation.
I profiled which function wastes time the most, and found it is `MapGrid::computeTargetDistance`.
This function has a `while` loop, which loops 1421744 times.
How could it be reduced, or what is the problem to make navigation slow down?

DLu, 2015-06-19:
Were you using a custom compiled version?

furushchev, 2015-06-19:
Now Yes, but it is almost the same with apt version of move_base

DLu, 2015-06-19:
How did you compile it?

furushchev, 2015-06-19:
with `catkin build move_base` on my catkin workspace inherited from `/opt/ros/hydro`

DLu, 2015-06-19:
http://answers.ros.org/question/71965/catkin-compiled-code-runs-3x-slower/

This should speed things up. If it does, please close the issue.

furushchev, 2015-06-19:
Ok, I'll test from now on. But, The speed was still slow even when I used move_base which I installed by `apt-get`

furushchev, 2015-06-19:
@DLu I tested by rebuilding move_base and related packages:

```
catkin clean -a
catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release
catkin build
```

It works! It made faster about by 2x-3x, but still it is too slow not to cause choppy motion. (it takes 0.34-0.95 sec. at each control loops)
Any idea?

furushchev, 2015-06-23:
At last, I solved. I have added StaticMap Plugin for local planner, which expanded GridMap size terribly and it increases loop counts of  computing target distances.
So, I remove StaticMap plugin from local planner (this plugin is actually also used as global planner), and checked its loop counts is now about 2500 times, which is very fast.
Anyway, @DLu  Thank you very much for kind advice.

closed 2015-06-23 by furushchev