ros-planning/navigation #372: robot_pose_ekf  odom_combined

Joylim, 2015-07-10:
when I run roslaunch turtlebot_bringup minial.launch  and  roslaunch robot_pose_ekf, 
we can see the topics /odom and /robot_pose_ekf both. But when I get the topic data ,
I find they are the same! I don't konw why and who can tell me? Thanks!

mkoledoye, 2015-07-10:
You should check the launch file for robot_pose_ekf. Or better put, do you have a launch file? (the command "roslaunch robot_pose_ekf" should have  a launch file appended at the end).
Are you remapping the filtered odometry data to /odom, and are there any other sensors (e.g. IMU) being used with the ekf?

mikeferguson, 2016-01-12:
this should either be on ros answers, or probably under turtlebot as that is where the configuration actually happens

closed 2016-01-12 by mikeferguson