ros-planning/navigation #373: NavFn Test - Missing Header

DLu, 2015-07-13:
```
$ catkin_make tests
...
/home/dlu/Catkin/jade_nav/src/navigation/navfn/src/read_pgm_costmap.cpp:40:17: fatal error: pgm.h: No such file or directory
 #include <pgm.h>
                 ^
compilation terminated.
...
```

mikeferguson, 2015-07-13:
@DLu -- i think this is actually not an issue with the code, but with your workflow -- did you ever do rosdep install on your workspace? navfn has a build_depend on "netpbm" (which it notes is actually a test depend) -- but no run depend, so you can't get it installed via installing the navigation debs.

DLu, 2015-07-13:
Sorry. New 14.04 machine.

closed 2015-07-13 by DLu

mikeferguson, 2015-07-13:
Not a problem -- I only knew the answer right off because I've hit it before.