ros-planning/navigation #376: ::getSizeInMetersX() and ::getSizeInMetersY() confusing

theidealist, 2015-07-14:
The implementation of ::getSizeInMetersX() and ::getSizeInMetersY() has been as follows for the entire history of costmap_2d.cpp's lifetime on github:

```
double Costmap2D::getSizeInMetersX() const
{
  return (size_x_ - 1 + 0.5) * resolution_;
}

double Costmap2D::getSizeInMetersY() const
{
  return (size_y_ - 1 + 0.5) * resolution_;
}
```

I must ask: why the `-1 + 0.5`?

I see the [documentation](http://docs.ros.org/jade/api/costmap_2d/html/classcostmap__2d_1_1Costmap2D.html#ad8986c48bda254f4c2d5db4f8b11150c) says
>The [x|y] size of the costmap (returns the centerpoint of the last legal cell in the map)

We can't understand how the centerpoint of the last legal cell in the map is the size of the map?  If we have a 10x10 costmap with resolution of 1.0, the size of that costmap is 9.5x9.5 meters?  I could *maybe* understand 9x9 if the measurement was centerpoint to centerpoint, but how/why does 9.5x9.5 make sense?

I propose the implementation be changed to (this shouldn't need a PR, should it?):
```
double Costmap2D::getSizeInMetersX() const
{
  return size_x_ * resolution_;
}

double Costmap2D::getSizeInMetersY() const
{
  return size_y_ * resolution_;
}
```

At the very least a further explanation in the documentation would really really help.

DLu, 2015-07-14:
I thought there was going to be a straight-forward answer to this, but I've got nothing. I apparently never questioned it. I tried to write out some rationale for it, but can't find it.

mikeferguson, 2015-07-14:
This does appear to be a bug. Perhaps the old costmap code used this "feature" -- but the new code definitely does not (full search is here https://github.com/ros-planning/navigation/search?utf8=%E2%9C%93&q=getSizeInMetersX).

When using a rolling costmap, it looks like our costmap is off by 1/2 of a cell resolution all the time.

Most interestingly clearCostmap is also probably incorrect because of this (https://github.com/ros-planning/navigation/blob/0d61d2adf21d6b71fd83edc5febbd6b781d76c9e/clear_costmap_recovery/src/clear_costmap_recovery.cpp#L154)

We certainly need to test a fix, but I think the proposed fix above is correct, and I cannot find any part of the main codebase that would be (negatively) affected.

AlexReimann, 2016-11-30:
I ran into this... this is madness .. bump

celynw, 2021-01-06:
Reporting from late 2020. The conclusion I came to is the same as @theidealist's fix. Now I find that it was reported in 2015.
Perosnally, I've been using the fix with no further problems, but it took me a while to track down this as the root of my problems in the first place.

closed 2025-06-04 by mikeferguson