ros-planning/navigation #379: MapServer: Image to OccupancyGrid conversion

DLu, 2015-07-29:
The map_server reads an image file and publishes a OccupancyGrid (OG). It reads the color of each pixel and gets the average intensity. 

The standard behavior is what I call a trinary map, where if the intensity is above the occ_thresh, the OG value is 100. If it is below the free_thresh, the OG value is 0. Otherwise the value is unknown, i.e. 255. 

Awhile back I added an additional option so that you could publish more values than that, such that if the value was between the two thresholds, it was linear interpolated between 0 and 100. This was not really documented on my part. Mea culpa. 

Now I have a use case where I'd like to use the full range of OG values 0-255. My plan was to add an additional mode where the intensity value is directly mapped to the OG value (since intensity is already 0-255). 

I already have code that does this. My questions for the community are
1) Does anyone have additional use cases for how the intensity to OG conversion is done
2) Does anyone use my hidden non-trinary option? Is there any value in keeping it around?

DLu, 2015-07-29:
https://github.com/DLu/navigation/tree/map_server_interpretation

jvgomez, 2015-10-04:
@DLu I think that to be able to load the map between 0 and 255 is a good idea as it is the more generic.

mikeferguson, 2016-02-13:
@DLu we're interested in this recently -- we actually just started using the non-trinary option, but as you say, it is not documented, and we'd actually prefer to have the raw version. One of my guys will probably create a PR next week based on an updated version of your branch.

DLu, 2020-07-03:
Hey this was actually fixed within the week. https://github.com/ros-planning/navigation/pull/422

closed 2020-07-03 by DLu