ros-planning/navigation #380: Installation Error

sj312931, 2015-08-10:
Hello,

I am trying to install the navigation package on my ubuntu 14.04 ROS-Indigo machine. When I try to build it I get the following error: 
/home/stv/test/src/navigation/map_server/src/image_loader.cpp:43:27: fatal error: SDL/SDL_image.h: No such file or directory
 #include <SDL/SDL_image.h>
                           ^
compilation terminated.

Can anyone help me with this? Is there a bug in the package?

sj312931, 2015-08-10:
This issue was resolved by installing the libsdl-image and libsdl-image-dev package. But I have one more problem now. 

/home/stv/test/src/navigation/move_base/include/move_base/move_base.h:46:43: fatal error: move_base_msgs/MoveBaseAction.h: No such file or directory
 #include <move_base_msgs/MoveBaseAction.h>
                                           ^
compilation terminated.

I am getting this error now. Can somebody help me with this?

lucbettaieb, 2015-08-10:
Try manually installing ros-indigo-move-base-msgs

sj312931, 2015-08-10:
@lucbettaieb Thanks for the reply. If manually install ros-indigo-move-base-msgs it doesnt give me any error. It works for me then. But if I use git clone https://github.com/ros-planning/navigation.git and try to build it then it gives me errors. Do you know the reason? 

What I mean is: 
I want to use the amcl package and I tried using git clone https://github.com/ros-planning/navigation.git to install the entire navigation package. I dont get errors if I manually install it but I get these errors when I clone the repository and use catkin_make to build the package.

lucbettaieb, 2015-08-10:
Try doing a rosdep install:
`rosdep install --from-paths src --ignore-src --rosdistro=indigo -y`

mikeferguson, 2015-08-10:
@lucbettaieb rosdep command is indeed what is needed -- there are build-only depends in map_server, so it won't be installed by installing debs and then building source on top, see also https://github.com/ros-planning/navigation/issues/373

closed 2015-08-10 by mikeferguson

lucbettaieb, 2015-08-10:
Also in the future please post these kinds of questions on ros answers.
answers.ros.org

sj312931, 2015-08-10:
Got it. Thanks for the help.

ESDA-LAB, 2018-08-26:
just install `apt-get install libmove-base-msgs-dev`