ros-planning/navigation #384: AMCL Minimal Update Distance

pengtang, 2015-08-17:
There is an parameter "update_min_d" in AMCL to specify the minimal update distance. Even though I set the parameter to be as small as 0.005, but the data I received indicates the minimal update distance is about 0.02. Is there any solution to this problem?

alittleharry, 2016-10-25:
maybe the change of odom between two message is 0.02 minimal.

mikeferguson, 2017-08-01:
Local testing shows that this parameter is working -- both from command line and through dynamic reconfigure -- I'm guessing the issue might be that you are setting the parameter in the wrong name space (the parameter is ~update_min_d, so it has to be in the namespace under the name of the amcl node (typically "amcl/update_min_d"))

closed 2017-08-01 by mikeferguson