ros-planning/navigation #400: Non-rectangular local costmaps

makokal, 2015-09-10:
Are there any plans to have non-rectangular local costmaps, which would reflect a more realistic state of real world robots? Something more of a semi-circle or a sector?

mikeferguson, 2015-09-10:
There are no such plans. Non-rectangular costmaps would require a significant change to the API given the way that updates propagate through layers.

DLu, 2015-09-10:
Such a change would change the fundamental data structure as well, in addition to the API. 

Besides the efficiency of carving off the edges of a square costmap into a circle, are there other benefits to changing the shape?

makokal, 2015-09-10:
@DLu 
Well, when designing specific behavior, say w.r.t people in the scene, I noticed standard local planners like `DWA` or `Eband`try to react as soon at the local cost map changes. When robot moves is curves, I notice sometimes my planner reacting a change that I would not want to (or that a circular local cost map would not take into account, like stuff much far behind). So it does affect behavior design.

DLu, 2015-09-10:
@makokal Just so I'm clear, are you seeing DWA change where it plans based on obstacles in the local costmap behind it?

makokal, 2015-09-11:
@DLu No, that is handled by DWA theory, maybe the example for not appropriate, rather my concern is is that I want to update local behavior (read local plan) when local cost map changes (but only in some region in the front). I suppose I can use additional layer to do this, but a customisable local cost map shape would make it easier.

DLu, 2015-09-11:
It seems as though it would be easier to write something that detected where changes happened within a rectangular layer rather than to change the underlying data structure.

makokal, 2015-09-11:
That is what I have so far, but needed to check plans just in case

mikeferguson, 2015-09-29:
Closing as non-rectangular local costmaps are not a feature we intend to implement

closed 2015-09-29 by mikeferguson