ros-planning/navigation #404: map_saver node bug

yukunlinykl, 2015-10-29:
the map_saver ignore the gray grid of map:

 71       fprintf(out, "P5\n# CREATOR: Map_generator.cpp %.3f m/pix\n%d %d\n255\n",
 72               map->info.resolution, map->info.width, map->info.height);
 73       for(unsigned int y = 0; y < map->info.height; y++) {
 74         for(unsigned int x = 0; x < map->info.width; x++) {
 75           unsigned int i = x + (map->info.height - y - 1) * map->info.width;
 76           if (map->data[i] == 0) { //occ [0,0.1)// <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<black
 77             fputc(254, out);
 78           } else if (map->data[i] == +100) { //occ (0.65,1]// <<<<<<<<<<<<<<<<<<<<<<<<<<<<white
 79             fputc(000, out);
 80           } else { //occ [0.1,0.65]//<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<gray
 81             fputc(205, out);// ????????????????????????????????????????????bug
 82           }
 83         }
 84       }

DLu, 2015-10-29:
This is not the ideal behavior for the map saver, but it does save the grey. Anything that is not 0 or 100 is saved as grey.

yukunlinykl, 2015-10-30:
how about save the map: [0,100]  to image: [rgb(0,0,0),rgb(255,255,255)] and map: -1 to image: (r!=g || g!=b || b!=r),
and change the map_server's behavior to support the image ?

if I fork a branch and modify the code as I say,
will your accept the merge request ?

DLu, 2015-11-01:
This changes the default behavior. map_server is used by many many people, and so we need to preserve the default behavior, perhaps by hiding it behind a flag.

yukunlinykl, 2015-11-02:
OK! thank you!

mikeferguson, 2016-02-13:
This is really a duplicate of https://github.com/ros-planning/navigation/issues/379

closed 2016-02-13 by mikeferguson