ros-planning/navigation #405: local_planner does not keep acc_lim in case of missed control loop

furushchev, 2015-11-09:
When we use `move_base` on our PR2 robot, PR2 often rattles. (= moves and suddenly stops repeatedly)
I found that the `realtime_loop` of PR2 subscribes `/cmd_vel`, moves its base along with the velocity, and when it does not receive commands about 200ms during moving, it suddenly stops for safety (equivalent to inserting velocity=0 ).

In ros wiki, the parameter `controller_frequency` is defined as below:

```
~controller_frequency (double, default: 20.0)
The rate in Hz at which to run the control loop and send velocity commands to the base.
```

and I set this parameter to `10.0`, but publish rate of commands are often lower than 4Hz.
(and it does not improve even if this parameter is `20.0` or `30.0`)

This means it breaks the `acc_lim` and causes sudden brake with huge twisting moment, which can cause damage to robot actuators.
I don't know this is a bug or not, but at least PR2 assumes that `move_base` publishes at least the rate set as `controller_frequency` in order to keep `acc_lim`.

I propose that if control loop cannot keep its rate, then `move_base` should publish velocity for deceleration keeping `acc_lim` set on param.
How do you think about it?

mikeferguson, 2015-11-09:
If you aren't getting the requested controller_frequency, then you are probably running out of CPU cycles. Are you using a stock pr2_navigation config, or have you made changes? What is the resolution of your map? Common issues that can cause this are having a map that is too fine of a resolution or configuring too many samples or too long of a simulation time for the DWA/trajectory_rollout planner.

DLu, 2015-12-09:
@furushchev Any updates?

furushchev, 2015-12-14:
@mikeferguson Thank you for advice, and I'm sorry for very late response.
We're now using DWA planner with `sim_time` set to 1.7 (default value)
We basically use `pr2_navigation` default configuration but changed `width` and `height` of local cost map from original 6.0 to 4.0.
After this change of cost map size, pr2 moves relatively more smoothly but still remains sudden stops. (the rate of `cmd_vel` becomes under about 5Hz despite of `controller_frequency` is set to 20.0)

Is there any other way to keep publishing `cmd_vel` at least not to break `acc_lim` set for safety?

mikeferguson, 2017-08-01:
In order to make the navigation stack work, it really needs to be tuned to keep the controller/planners working at the right frequency. Certainly, the default PR2 nav used to do this (I can't speak about it post-2014 or so). I know our Fetch robots (with a fraction of the processing power that a PR2 has) do a fine job on less than quarter of an i5 processor. If you really can't keep the controller frequency constant, I'd look deeper into what else might be running and starving the nav stack of CPU. How much CPU is nav using?

The reason we can't really "fix this" is that if the process is starved, the controller can't just "publish velocities that ensure acc_lim is met" since, well, it has no idea if/when it is going to have CPU cycles to actually publish the next message. Guaranteeing CPU cycles would likely require migrating parts of the controller into a real-time loop (which is also fairly untenable for this project to then be generic enough for so many ROS-based robots).

closed 2017-08-01 by mikeferguson