ros-planning/navigation #408: With Rolling Window on, costmap_2d not properly updating bounds and costs in the static layer

palmieri, 2015-12-11:
Hi All,

When setting `rolling_window` to true and `static_map` to false and having as loaded plugins a `Static Layer` , an `Obstacle Layer` and the `Inflation Layer`, the costmap_2d is not properly updating the bounds and the costs of the rolling map.

Example

Initial Map:
[image]

While Moving: 
[image]

Here the parameters used to launch the costmap:
[globalcostmap.txt](https://github.com/ros-planning/navigation/files/59338/globalcostmap.txt)

While rolling the costmap_2d is updating the costs only in a smaller window, the bounds are not properly updated.

To avoid the issue, the lines 261-2 of the file `costmap_2d/plugins/static_layer.cpp`
```c++
  if (!map_received_ || !(has_updated_data_ || has_extra_bounds_))
    return;
```

could be changed into:
```c++
  if( !layered_costmap_->isRolling() ){
    if ( !map_received_ || !(has_updated_data_ || has_extra_bounds_))
        return;
  }
```

I hope I didn't miss anything in the parameters that would solve this issue. If it is needed I can create a pull request for that.

Best,

Luigi

mikeferguson, 2015-12-20:
What is the range of your laser sensor -- relatively short compared to the size of your rolling map? Having looked through the code now, it looks like the "static_map" parameter no longer does anything if you specify your set of plugins (since it is only used by the legacy setup code when people do not specify plugins to use). Thus, your setup is the same as what I am typically using, but I have a 25m laser and a smaller rolling map (so the bounds for me are likely coming from the laser itself).

The suggested fix should certainly work -- although the more elegant solution is probably to override updateOrigin to do the actual static cost map copying to the local copy within the plugin and add updateOrigin to be done each update (as is the case with all the other plugins). If you want to open a PR for the suggested fix against Indigo and Jade branches I would certainly merge that. I'll try to find some time to implement the more "proper" fix so that static_map works closer to the way the other layers work (the rolling static map support is relatively new, only about 6 months old).

DLu, 2016-01-17:
Ready to close?

mikeferguson, 2016-01-17:
@DLu -- I think this has not been backported to indigo -- let me check on that before we close

mikeferguson, 2016-01-17:
#417 fixes this in indigo

closed 2016-01-17 by mikeferguson