ros-planning/navigation #409: There is no clear_unknown_space service

sloretz, 2015-12-15:
The jade documentation claims there is a service called ~clear_unknown_space in section 1.1.5
http://wiki.ros.org/move_base

It appears this service doesn't actually exist. "rosservice list | grep unknown" shows no such service, and searching this repository for clear_unknown_space returns 0 results.

DLu, 2016-01-17:
This was removed in Hydro (probably because I didn't see the use for it). What would you like to see happen?

1) The documentation updated to reflect the current state?
2) The service reintroduced to function like described?

sloretz, 2016-02-16:
I would be fine with just the documentation being updated, and I think I see a use for this service.
```
                                     (goal)
 #1                       #2

+---------+        +     +---------+        +
|         |        |     |         |        |
|         | (Real  |     |         |(Phantom|
| (goal)  |  Dog)  |     | (Robot) | Dog)   |
|         |        |     |         |        |
|         |        |     |         |        |
|         |        |     |         |        |
|         |        |     |         |        |
|         |        |     |         |        |   (Real
|         |(Robot) |     |         |        |    Dog)
|         |        |     |         |        |
|         +        |     |         +        |
|                  |     |                  |
|                  |     |                  |
|                  |     |                  |
|                  |     |                  |
+------------------+     +------------------+
```

Imagine a robot with an obstacle layer on the global costmap navigates into a tight U shaped space, seeing a dog block the path behind it before moving around the bend. Later after the dog has moved away, the robot is commanded to move to a goal outside of the U. The global planner will fail to produce a path because the dog is still in the global costmap. The default clear costmap recovery behaviors won't remove the phantom dog because it is physically close to the robot.

I've observed that if the costmap is cleared and a goal is sent immediately, the robot may move with an empty costmap and collide with the walls.

The clear_unknown_space service could remove the phantom dog without removing the robot's knowledge of the walls.

sloretz, 2016-02-17:
Ooops looks like I misunderstood the documentation. I see now that clear unknown space clears cells with the cost value "uknown". I expected it would remove cells that the robot couldn't see directly, meaning their current costs were unknown.

Is there an existing recovery plugin that can clear costmap cells that haven't been updated in a while?

I think with a documentation update this issue can be closed. Thanks!

DLu, 2016-03-03:
I started looking into your use case last year, but didn't write too robust of a solution for it. https://github.com/DLu/decaying_obstacle_layer

You should be able to clear the obstacle with a clear_costmap with distance = 0.

I've updated the documentation to reflect the fact that the service was removed.

closed 2016-03-03 by DLu