ros-planning/navigation #421: `dwa_local_planner` is struggling at the near to end of the path

orhangazi44, 2016-02-11:
After updating ros packages from ubuntu repositories today, `dwa_local_planner` is trying to escape infinitely with struggling with little movements at near to end of the path, changing `path_distance_bias` and `goal_distance_bias` parameters doesn't make sense:

[Screenshot at struggling](http://answers.ros.org/upfiles/14551904439069127.png)
`[ INFO] [1455190327.742058584, 6646.566000000]: Got new plan`
`[ INFO] [1455190327.908304049, 6646.664000000]: Got new plan`
`[ INFO] [1455190328.263322957, 6646.864000000]: Got new plan`
`[ INFO] [1455190328.617373718, 6647.064000000]: Got new plan`
`[ INFO] [1455190328.925895750, 6647.264000000]: Got new plan`
`[ INFO] [1455190329.222412376, 6647.465000000]: Got new plan`

You can read more information in this [ROS Answers Thread].(http://answers.ros.org/question/226231/dwa-planner-follows-global-path-closely-and-shortening-path-in-some-cases/)

mkoledoye, 2016-02-12:
@orhangazi44 To be sure I understand you correctly, the robot makes small rotational movements just before reaching the target goal? If that's the case, trying increasing the controller_frequency of the planner. if that does not fix it, try setting the xy_goal_tolerance and yaw_goal_tolerance to their defaults. Had the same problem of recent, and these happened to be my fix.

orhangazi44, 2016-02-12:
Thank you bro!!!! Doing both things helped. But I think It's still an issue. I setted `controller_frequency` to `1.0`. 5.0, 10.0 not worked well. And setted tolerance parameters to default. Now it tries different paths and reaching the goal. Thank you again.

orhangazi44, 2016-03-15:
Closing, because there is not enough information, I'll open new one.

closed 2016-03-15 by orhangazi44