ros-planning/navigation #430: Nav stack release 1.12.6 breaks repopulation of static layer on costmap clear

mirzashah, 2016-03-04:
@alexhenning @mikeferguson 

The changes in 1.12.6 created a regression for us. When the costmap is cleared using the clear costmap service exposed by move_base, the static layer does not repopulate with the current map. This resulted in some interesting behavior where the robot wants to drive through walls.

https://github.com/ros-planning/navigation/commit/31accb2e35f0506c3dae08315da1583bb6761403

Reverting back to 1.12.5 appears to fix this.

RaymondKYLiu, 2016-03-09:
This condition block it.
`if` (map_sub_.getTopic() != ros::names::resolve(map_topic))

corot, 2016-04-13:
Same problem for me. An easy workaround is to set first_map_only static layer parameter as true. This makes the static layer keep a copy of the first map received and use it on resets. It shutdowns the map subscriber to ensure it only uses this stored copy.

This parameter (and many others) is not documented on the wiki since current layer pages date from hydro! (I opened an issue #439)

alexhenning, 2016-04-25:
@mirzashah Could you post the full output of `rosparam get /move_base`

mirzashah, 2016-04-26:
@alexhenning is it ok if the output comes from 1.12.5 or do you need it specifically from 1.12.6?

alexhenning, 2016-04-28:
@mirzashah I assume you're using the same parameters with both? If the output differs, then that should be unrelated to the source change.

alexhenning, 2016-05-04:
@mirzashah https://github.com/ros-planning/navigation/pull/446 should fix this bug and be in the next release. Can you verify that it resolves your problem.

mikeferguson, 2016-05-26:
446 is released into 1.12.8 -- we've not gotten a yes/no from original reporters -- but I'm going to close this. Please reopen if you find that the fix does not work and can provide additional info

closed 2016-05-26 by mikeferguson

corot, 2016-05-30:
Fixed!   thanks

mirzashah, 2016-06-07:
@mikeferguson @alexhenning  Thanks so much for the fix --  just tested with 1.12.10, regression seems to be fixed

mikeferguson, 2016-06-08:
@mirzashah -- awesome -- and apologies on that -- we really did test the crap out of this stuff and somehow never managed to get it into that state.