ros-planning/navigation #432: Global Costmap is not static anymore (although static is enabled)

ssrselvamraju, 2016-03-12:
After the upgrade to Navstack 1.12.7 from 1.12.4, we're now experiencing the following issues:

1. In 1.12.4 I didn't have to specify the width and height of the global costmap, and it was automatically picking up to be the size of the actual "map". After the 1.12.7 upgrade seems like I'm having to set the height and width (discovered this, when the global costmap was defaulting to size of 10x10 after the navstack upgrade)
2. Although the height and width is set to 10,000x10,000 the size seems to be shrinking based on the distance the robot is travelling, so if there's a goal position set to a different point in the map which is now not in the global costmap, the robot is unable to plan there.

Global costmap with navstack 1.12.7 (and height X width = 10,000 x 10,000)

[image]

Global costmap with navstack 1.12.4 (and height X width is not explicitly set anywhere, but default params are 10x10. Also publish frequency is 0, since this is a static layer and doesn't have to update)

[image]

Any suggestions/solutions for me to try?

mirzashah, 2016-03-12:
I believe you're experiencing same issue I had #430 . I fell back to 1.12.5, the code breaks in 1.12.6

mikeferguson, 2016-03-12:
Do either of you have a link to your navigation configuration? We're not seeing this issue in house, so I'm hoping to narrow down what you might be using that is different from our config.

mikeferguson, 2016-03-12:
Also, a trace of the ROSOUT would be useful, in particular, if you see "Requesting the map..." or "Received a map.."

ssrselvamraju, 2016-03-15:
$ cat costmap_common_params.yaml 
```
obstacle_range: 3.0
raytrace_range: 5.0
robot_radius: 0.5
inflation_radius: 0.5

observation_sources: velodyne_scan laser_scan_sensor laser_scan_sensor2 #laser_scan_sensor3

velodyne_scan: {sensor_frame: velodyne, data_type: LaserScan, topic: velodyne_scan, marking: true, clearing: true}

laser_scan_sensor: {sensor_frame: LMS151, data_type: LaserScan, topic: bottom_scan, marking: true, clearing: true}

laser_scan_sensor2: {sensor_frame: TiM551, data_type: LaserScan, topic: top_scan, marking: true, clearing: true}

#laser_scan_sensor3: {sensor_frame: LMS111, data_type: LaserScan, topic: back_scan, marking: true, clearing: true}

```
$ cat global_costmap_params.yaml 
```
global_costmap:
  map_type: costmap
  global_frame: map
  robot_base_frame: base_link
  update_frequency: 5.0
  static_map: true
  track_unknown_space: true
  unknown_cost_value: 253

```
$ cat local_costmap_params.yaml 
```
local_costmap:
  map_type: costmap
  global_frame: map
  robot_base_frame: base_link
  update_frequency: 10.0
  publish_frequency: 5.0
  static_map: false
  rolling_window: true
  width: 8.0
  height: 8.0
  resolution: 0.05

  cost_scaling_factor: 0.0

```

ssrselvamraju, 2016-03-15:
A few traces of the logs:

First some grep:
```
~/.ros/log/b4f07922-e7ea-11e5-974e-78d00421308d$ grep "Requesting the map" *
move_base_delay-60-stdout.log:[ INFO] [1457743138.107876403]: Requesting the map...
move_base_delay-60-stdout.log:[ INFO] [1457989235.181643366]: Requesting the map...
rosout.log.1:1457989235.181657062 INFO [/navigation_stack/navigation-latest/costmap_2d/plugins/static_layer.cpp:87(StaticLayer::onInitialize) [topics: /rosout, /cmd_vel, /move_base/current_goal, /move_base/goal] Requesting the map...

:~/.ros/log/b4f07922-e7ea-11e5-974e-78d00421308d$ grep "Received a map" *

:~/.ros/log/b4f07922-e7ea-11e5-974e-78d00421308d$ grep "Received" *
move_base_delay-60-stdout.log:[ INFO] [1457743138.413897370]: Received a 1300 X 1300 map at 0.050000 m/pix
move_base_delay-60-stdout.log:[ INFO] [1457743137.033320141]: Received a 1300 X 1300 map @ 0.050 m/pix
move_base_delay-60-stdout.log:[ INFO] [1457989235.486678993]: Received a 1300 X 1300 map at 0.050000 m/pix
rosout.log.1:1457989235.486699380 INFO [/src/k5_navigation_stack/navigation-latest/costmap_2d/plugins/static_layer.cpp:99(StaticLayer::onInitialize) [topics: /rosout, /cmd_vel, /move_base/current_goal, /move_base/goal] Received a 1300 X 1300 map at 0.050000 m/pix
```

Here are some traces from the *move_base_delay-60-stdout.log*:

```
 * /move_base/loc^[[33m[ WARN] [1457743127.982014340]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: . canTransform returned after 0.10072 timeout was 0.1.^[[0m
^[[33m[ WARN] [1457743133.023318881]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: . canTransform returned after 0.100802 timeout was 0.1.^[[0m
^[[0m[ INFO] [1457743137.953746523]: Loading from pre-hydro parameter style^[[0m
^[[0m[ INFO] [1457743138.026017275]: Using plugin "static_layer"^[[0m
^[[0m[ INFO] [1457743138.107876403]: Requesting the map...^[[0m
^[[0m[ INFO] [1457743138.316194093]: Resizing costmap to 1300 X 1300 at 0.050000 m/pix^[[0m
^[[0m[ INFO] [1457743138.413897370]: Received a 1300 X 1300 map at 0.050000 m/pix^[[0m
^[[0m[ INFO] [1457743138.426256892]: Using plugin "obstacle_layer"^[[0m
^[[0m[ INFO] [1457743138.432730816]:     Subscribed to Topics: velodyne_scan laser_scan_sensor laser_scan_sensor2^[[0m
^[[0m[ INFO] [1457743138.545825845]: Using plugin "inflation_layer"^[[0m
^[[0m[ INFO] [1457743138.761383309]: Loading from pre-hydro parameter style^[[0m
^[[0m[ INFO] [1457743138.790227503]: Using plugin "obstacle_layer"^[[0m
^[[0m[ INFO] [1457743138.851336549]:     Subscribed to Topics: velodyne_scan laser_scan_sensor laser_scan_sensor2^[[0m
^[[0m[ INFO] [1457743138.914020751]: Using plugin "inflation_layer"^[[0m
^[[0m[ INFO] [1457743139.046973969]: Created local_planner base_local_planner/TrajectoryPlannerROS^[[0m
^[[0m[ INFO] [1457743139.060398137]: Sim period is set to 0.10^[[0m
^[[33m[ WARN] [1457743139.079141598]: You've specified a positive escape velocity. This is probably not what you want and will cause the robot to move forward instead of backward. You should probably change your escape_vel parameter to be negative^[[0m
^[[33m[ WARN] [1457743141.117449821]: Map update loop missed its desired rate of 5.0000Hz... the loop actually took 2.4833 seconds^[[0m
^[[0m[ INFO] [1457743141.228343349]: Recovery behavior will clear layer obstacles^[[0m
^[[0m[ INFO] [1457743141.295915976]: Recovery behavior will clear layer obstacles^[[0m
^[[33m[ WARN] [1457743143.549180791]: Map update loop missed its desired rate of 5.0000Hz... the loop actually took 2.4317 seconds^[[0m
^[[0m[ INFO] [1457743143.584887743]: odom received!^[[0m
^[[33m[ WARN] [1457743332.059942358]: The origin for the sensor at (55.48, 14.19) is out of map bounds. So, the costmap cannot raytrace for it.^[[0m
^[[33m[ WARN] [1457743561.880260307]: Costmap2DROS transform timeout. Current time: 1457743561.8801, global_pose stamp: 1457743561.5727, tolerance: 0.3000^[[0m
^[[33m[ WARN] [1457743561.880338334]: Could not get robot pose, cancelling reconfiguration^[[0m
^[[33m[ WARN] [1457743562.915057575]: Costmap2DROS transform timeout. Current time: 1457743562.9150, global_pose stamp: 1457743561.5727, tolerance: 0.3000^[[0m
^[[33m[ WARN] [1457743562.915153212]: Could not get robot pose, cancelling reconfiguration^[[0m
^[
```
...

```
[[33m[ WARN] [1457809496.600207867]: Map update loop missed its desired rate of 10.0000Hz... the loop actually took 0.1998 seconds^[[0m
^[[33m[ WARN] [1457809496.711431632]: Control loop missed its desired rate of 10.0000Hz... the loop actually took 0.1230 seconds^[[0m
^[[33m[ WARN] [1457809496.809094271]: Control loop missed its desired rate of 10.0000Hz... the loop actually took 0.1207 seconds^[[0m
^[[33m[ WARN] [1457809496.919299675]: Control loop missed its desired rate of 10.0000Hz... the loop actually took 0.1309 seconds^[[0m
^[[33m[ WARN] [1457809496.919743820]: Map update loop missed its desired rate of 10.0000Hz... the loop actually took 0.4194 seconds^[[0m
^[[33m[ WARN] [1457809497.017091350]: Control loop missed its desired rate of 10.0000Hz... the loop actually took 0.1287 seconds^[[0m
^[[33m[ WARN] [1457809497.125917877]: Control loop missed its desired rate of 10.0000Hz... the loop actually took 0.1375 seconds^[[0m
^[[33m[ WARN] [1457809497.126483342]: Map ^[[0m[ INFO] [1457743136.828834422]: Subscribed to map topic.^[[0m
^[[0m[ INFO] [1457743137.033320141]: Received a 1300 X 1300 map @ 0.050 m/pix
^[[0m
[... cut ...]
^[[33m[ WARN] [1457989271.438769466]: Map update loop missed its desired rate of 5.0000Hz... the loop actually took 0.4083 seconds^[[0m
^[[33m[ WARN] [1457989271.742303751]: Map update loop missed its desired rate of 5.0000Hz... the loop actually took 0.3035 seconds^[[0m
^[[33m[ WARN] [1457989272.038628930]: Map update loop missed its desired rate of 5.0000Hz... the loop actually took 0.3999 seconds^[[0m
^[[33m[ WARN] [1457989272.326183762]: Map update loop missed its desired rate of 5.0000Hz... the loop actually took 0.4874 seconds^[[0m
^[
```

I can provide the full move-base log if needed. It's pretty big though.

ssrselvamraju, 2016-03-18:
Agree to @mirzashah's comment. The issue is found in 1.12.6+ and works fine until 1.12.5.

alexhenning, 2016-04-25:
@ssrselvamraju Could you post the pull output of `rosparam get /move_base`

ssrselvamraju, 2016-04-27:
@alexhenning Here you go:

`rosparam  get /move_base/
NavfnROS: {allow_unknown: false}
TrajectoryPlannerROS: {acc_lim_th: 9.0, acc_lim_theta: 3.2, acc_lim_x: 3.0, acc_lim_y: 2.5,
  angular_sim_granularity: 0.025, dwa: false, escape_reset_dist: 0.1, escape_reset_theta: 1.57079632679,
  escape_vel: 0.0, gdist_scale: 0.8, heading_lookahead: 0.325, heading_scoring: false,
  heading_scoring_timestep: 0.1, holonomic_robot: false, max_rotational_vel: 0.5,
  max_vel_theta: 1.0, max_vel_x: 0.6, meter_scoring: true, min_in_place_rotational_vel: 0.2,
  min_in_place_vel_theta: 0.4, min_vel_theta: -1.0, min_vel_x: 0.25, occdist_scale: 0.01,
  oscillation_reset_dist: 0.05, pdist_scale: 1.2, restore_defaults: false, sim_granularity: 0.025,
  sim_time: 1.7, simple_attractor: false, vtheta_samples: 20, vx_samples: 20, xy_goal_tolerance: 0.2,
  y_vels: '-0.3,-0.1,0.1,-0.3', yaw_goal_tolerance: 0.25}
aggressive_reset: {reset_distance: 1.84}
base_global_planner: navfn/NavfnROS
base_local_planner: base_local_planner/TrajectoryPlannerROS
clearing_rotation_allowed: true
conservative_reset: {reset_distance: 3.0}
conservative_reset_dist: 3.0
controller_frequency: 10.0
controller_patience: 5.0
global_costmap:
  footprint: '[]'
  footprint_padding: 0.01
  global_frame: map
  height: 10
  inflation_layer: {cost_scaling_factor: 10.0, enabled: true, inflation_radius: 1.6}
  map_cloud: {clearing: true, data_type: PointCloud2, marking: true, sensor_frame: base_link,
    topic: map_pointcloud}
  map_type: costmap
  obstacle_layer:
    combination_method: 1
    enabled: true
    footprint_clearing_enabled: true
    laser_scan_sensor: {clearing: true, data_type: LaserScan, marking: true, sensor_frame: LMS151,
      topic: bottom_scan}
    laser_scan_sensor2: {clearing: true, data_type: LaserScan, marking: true, sensor_frame: TiM551,
      topic: top_scan}
    max_obstacle_height: 2.0
    observation_sources: velodyne_scan laser_scan_sensor laser_scan_sensor2
    obstacle_range: 3.0
    raytrace_range: 5.0
    track_unknown_space: true
    velodyne_scan: {clearing: true, data_type: LaserScan, marking: true, sensor_frame: velodyne,
      topic: velodyne_scan}
  origin_x: 0.0
  origin_y: 0.0
  plugins:
  - {name: static_layer, type: 'costmap_2d::StaticLayer'}
  - {name: obstacle_layer, type: 'costmap_2d::ObstacleLayer'}
  - {name: inflation_layer, type: 'costmap_2d::InflationLayer'}
  publish_frequency: 0.0
  resolution: 0.05
  robot_base_frame: base_link
  robot_radius: 0.5
  static_layer: {enabled: true, track_unknown_space: true, unknown_cost_value: 253}
  static_map: true
  transform_tolerance: 0.3
  update_frequency: 5.0
  width: 10
local_costmap:
  footprint: '[]'
  footprint_padding: 0.01
  global_frame: map
  height: 8
  inflation_layer: {cost_scaling_factor: 0.0, enabled: true, inflation_radius: 1.6}
  map_cloud: {clearing: true, data_type: PointCloud2, marking: true, sensor_frame: base_link,
    topic: map_pointcloud}
  map_type: costmap
  obstacle_layer:
    combination_method: 1
    enabled: true
    footprint_clearing_enabled: true
    laser_scan_sensor: {clearing: true, data_type: LaserScan, marking: true, sensor_frame: LMS151,
      topic: bottom_scan}
    laser_scan_sensor2: {clearing: true, data_type: LaserScan, marking: true, sensor_frame: TiM551,
      topic: top_scan}
    max_obstacle_height: 2.0
    observation_sources: velodyne_scan laser_scan_sensor laser_scan_sensor2
    obstacle_range: 3.0
    raytrace_range: 5.0
    velodyne_scan: {clearing: true, data_type: LaserScan, marking: true, sensor_frame: velodyne,
      topic: velodyne_scan}
  origin_x: 0.0
  origin_y: 0.0
  plugins:
  - {name: obstacle_layer, type: 'costmap_2d::ObstacleLayer'}
  - {name: inflation_layer, type: 'costmap_2d::InflationLayer'}
  publish_frequency: 5.0
  resolution: 0.05
  robot_base_frame: base_link
  robot_radius: 0.5
  rolling_window: true
  static_map: false
  transform_tolerance: 0.3
  update_frequency: 10.0
  width: 8
oscillation_distance: 0.5
oscillation_timeout: 0.0
planner_frequency: 0.5
planner_patience: 5.0
recovery_behavior_enabled: false
restore_defaults: false
shutdown_costmaps: false`

alexhenning, 2016-04-28:
@ssrselvamraju ran the following launch file on one of our robots with 1.12.6 installed from debs and a source checkout of 1.12.7 and I was unable to reproduce it. What are you using to serve the map? Is it possible that something has changed with that? The best way to move forward be to use a source check and run [git bisect](https://git-scm.com/docs/git-bisect) to try to isolate the commit that breaks things and we can look at that. In terms of me reproducing things, it would be helpful if you could provide more details on exactly what you're running and steps to reproduce the problem.

move_base.launch
```xml
<launch>
  <!-- set topics -->
  <arg name="map_topic" default="map" />
  <arg name="cmd_vel_topic" default="cmd_vel" />
  <arg name="odom_topic" default="odom" />

  <!-- planners, costmaps, etc -->
  <node pkg="move_base" type="move_base" name="move_base" respawn="true" output="screen">
    <rosparam file="path/to/config.yaml" command="load" ns="/" />
    <remap from="cmd_vel" to="$(arg cmd_vel_topic)"/>
    <remap from="odom" to="$(arg odom_topic)"/>
    <remap from="map" to="$(arg map_topic)" />
  </node>
</launch>
```
config.yaml
```yaml
move_base:
  NavfnROS: {allow_unknown: false}
  TrajectoryPlannerROS: {acc_lim_th: 9.0, acc_lim_theta: 3.2, acc_lim_x: 3.0, acc_lim_y: 2.5,
  angular_sim_granularity: 0.025, dwa: false, escape_reset_dist: 0.1, escape_reset_theta: 1.57079632679,
  escape_vel: 0.0, gdist_scale: 0.8, heading_lookahead: 0.325, heading_scoring: false,
  heading_scoring_timestep: 0.1, holonomic_robot: false, max_rotational_vel: 0.5,
  max_vel_theta: 1.0, max_vel_x: 0.6, meter_scoring: true, min_in_place_rotational_vel: 0.2,
  min_in_place_vel_theta: 0.4, min_vel_theta: -1.0, min_vel_x: 0.25, occdist_scale: 0.01,
  oscillation_reset_dist: 0.05, pdist_scale: 1.2, restore_defaults: false, sim_granularity: 0.025,
  sim_time: 1.7, simple_attractor: false, vtheta_samples: 20, vx_samples: 20, xy_goal_tolerance: 0.2,
  y_vels: '-0.3,-0.1,0.1,-0.3', yaw_goal_tolerance: 0.25}
  aggressive_reset: {reset_distance: 1.84}
  base_global_planner: navfn/NavfnROS
  base_local_planner: base_local_planner/TrajectoryPlannerROS
  clearing_rotation_allowed: true
  conservative_reset: {reset_distance: 3.0}
  conservative_reset_dist: 3.0
  controller_frequency: 10.0
  controller_patience: 5.0
  global_costmap:
    footprint: '[]'
    footprint_padding: 0.01
    global_frame: map
    height: 10
    inflation_layer: {cost_scaling_factor: 10.0, enabled: true, inflation_radius: 1.6}
    map_cloud: {clearing: true, data_type: PointCloud2, marking: true, sensor_frame: base_link,
      topic: map_pointcloud}
    map_type: costmap
    obstacle_layer:
      combination_method: 1
      enabled: true
      footprint_clearing_enabled: true
      laser_scan_sensor: {clearing: true, data_type: LaserScan, marking: true, sensor_frame: LMS151,
        topic: bottom_scan}
      laser_scan_sensor2: {clearing: true, data_type: LaserScan, marking: true, sensor_frame: TiM551,
        topic: top_scan}
      max_obstacle_height: 2.0
      observation_sources: velodyne_scan laser_scan_sensor laser_scan_sensor2
      obstacle_range: 3.0
      raytrace_range: 5.0
      track_unknown_space: true
      velodyne_scan: {clearing: true, data_type: LaserScan, marking: true, sensor_frame: velodyne,
        topic: velodyne_scan}
    origin_x: 0.0
    origin_y: 0.0
    plugins:
    - {name: static_layer, type: 'costmap_2d::StaticLayer'}
    - {name: obstacle_layer, type: 'costmap_2d::ObstacleLayer'}
    - {name: inflation_layer, type: 'costmap_2d::InflationLayer'}
    publish_frequency: 0.0
    resolution: 0.05
    robot_base_frame: base_link
    robot_radius: 0.5
    static_layer: {enabled: true, track_unknown_space: true, unknown_cost_value: 253}
    static_map: true
    transform_tolerance: 0.3
    update_frequency: 5.0
    width: 10
  local_costmap:
    footprint: '[]'
    footprint_padding: 0.01
    global_frame: map
    height: 8
    inflation_layer: {cost_scaling_factor: 0.0, enabled: true, inflation_radius: 1.6}
    map_cloud: {clearing: true, data_type: PointCloud2, marking: true, sensor_frame: base_link,
      topic: map_pointcloud}
    map_type: costmap
    obstacle_layer:
      combination_method: 1
      enabled: true
      footprint_clearing_enabled: true
      laser_scan_sensor: {clearing: true, data_type: LaserScan, marking: true, sensor_frame: LMS151,
        topic: bottom_scan}
      laser_scan_sensor2: {clearing: true, data_type: LaserScan, marking: true, sensor_frame: TiM551,
        topic: top_scan}
      max_obstacle_height: 2.0
      observation_sources: velodyne_scan laser_scan_sensor laser_scan_sensor2
      obstacle_range: 3.0
      raytrace_range: 5.0
      velodyne_scan: {clearing: true, data_type: LaserScan, marking: true, sensor_frame: velodyne,
        topic: velodyne_scan}
    origin_x: 0.0
    origin_y: 0.0
    plugins:
    - {name: obstacle_layer, type: 'costmap_2d::ObstacleLayer'}
    - {name: inflation_layer, type: 'costmap_2d::InflationLayer'}
    publish_frequency: 5.0
    resolution: 0.05
    robot_base_frame: base_link
    robot_radius: 0.5
    rolling_window: true
    static_map: false
    transform_tolerance: 0.3
    update_frequency: 10.0
    width: 8
  oscillation_distance: 0.5
  oscillation_timeout: 0.0
  planner_frequency: 0.5
  planner_patience: 5.0
  recovery_behavior_enabled: false
  restore_defaults: false
  shutdown_costmaps: false
```

```
shell1 $ roslaunch move_base.launch
shell2 $ roslaunch map # Either map server or slam
shell3 $ rosservice call /move_base/clear_costmaps "{}"  # Called lots of times
```

ssrselvamraju, 2016-04-29:
@alexhenning Can you give me some information on how long you ran the above test for?

Also we use map_server to serve the map:
`
 <node name="map_server" pkg="map_server" type="map_server" args="$(find robot_2dnav)/map.pgm 0.05"/> `

The issue usually appears after several hours of the robot running. Opening closing rviz multiple times, and viewing the 'global_costmap' on rviz. I usually can see the issue after running our robot overnight.

I'll do a git bisect and see what's up, and will post details when I get a lead.

alexhenning, 2016-04-29:
@ssrselvamraju I only ran with your config for half an hour. We've had many robots running with our config for long periods of time without noticing a similar problem, which would continue to hint that it might be a parameter interaction. One thing that we could try to speed up reproduction is to create a launch file that launches move_base, map_server, static tf publisher, and a custom node that calls clear costmaps multiple times a minute. If it is either a costmap threading or timing related bug, this may significantly speed up reproducing the issue.

alexhenning, 2016-05-04:
@ssrselvamraju https://github.com/ros-planning/navigation/pull/446 should fix this bug and will be in the next release. Can you verify that you can't reproduce the problem with it?

asimay, 2016-05-24:
I came across the same issues... it is a dead loop whatever i adjust the parameters. 
it seems that your local map's width & height must be small than 5m, if width&height >5, it will occur the issue. but i used the sick lms's range more than 50m..
[image]

asimay, 2016-05-24:
I'm glad to test this issue with your new changes and feedback to you.

asimay, 2016-05-25:
hi, @alexhenning , it's a very bad experience.... I download the new code, the fix code #446 and all code changes in new git version still didn't solve the problem....
please check the screenshot below, thanks!

[image]

asimay, 2016-05-25:
I test in stage and real env. same issue.
with global costmap:   width: 6.0,   height: 6.0
[image]

asimay, 2016-05-25:
if set global costmap: width: 4.0, height: 4.0, less than 5m as i said above, (all related modification), the loop warning disappeared...

asimay, 2016-05-25:
and my computer : 2 cpu, 12 core, intel E5, 2.6Ghz, with 32G RAM, 64bit win7.

asimay, 2016-05-25:
I think maybe the map calculation with too long time,
 in costmap_2d_ros.cpp     ROS_WARN("Map update loop missed its desired rate of %.4fHz... the loop actually took %.4f seconds", frequency,

maybe use thread or optimise the program to speed up the calculation time.

asimay, 2016-05-25:
hi, all, I think I figure out the root cause:
in costmap_2d_ros.cpp ROS_WARN("Map update loop missed its desired rate of %.4fHz... the loop actually took %.4f seconds", frequency,
line 409,
r.sleep();
if (r.cycleTime() > ros::Duration(1 / frequency)) , the loop time is wrong,
in ros wiki it said: 
Duration 	cycleTime ()
 	Get the actual run time of a cycle from start to sleep. 
so the cycletime is loop time +  ros::Duration(1 / frequency), it is of cause bigger than  ros::Duration(1 / frequency),
I test the map loop time in function, it is actually very small, and can satisfy our needs.
if we comment the r.sleep();
the output is good.
and with the same method, we can remove the controller loop warning in the same way.
it acts good finally.

let you all know. thanks!

alexhenning, 2016-05-25:
@asimay That looks like a performance issue with your machine, not necessarily a bug in costmaps. Since you say you're running Windows 7, but you have screenshots of an Ubuntu terminal, I assume you're running in a VM? If that is correct, I would make sure that you're allocating enough CPU and RAM to the VM. If that doesn't resolve your issue, please post more details in new ticket since this would be separate from the bug @ssrselvamraju was experiencing.

asimay, 2016-05-26:
dear @alexhenning, sorry, i make a typo mistake. i'm running ubuntu, and computer with very good performance, enough CPU and RAM, it is a powerful Dell workstation.
frome the code, if we comment the r.sleep(), the warning is never occured, so it proved the code real cycletime is smaller than frequency time, so, there is some error in loop time calculation(here is r.cycletime() ). I think we can open a new ticket and verify, because really many people met this issue, and i found lots of information with no result, so i checked the code and found that.

mikeferguson, 2016-05-26:
446 is released into 1.12.8 -- we've not gotten a yes/no from original reporters -- but I'm going to close this since it is getting full of completely un-related reports. Please reopen if you find that the fix does not work and can provide additional info.

closed 2016-05-26 by mikeferguson

HappySamuel, 2021-11-03:
Hi @alexhenning 

I discovered that whenever you rosservice call /move_base/clear_costmap, it will cause the map update loop missed its desired rate. Do you have this issue when using a custom node to constantly clearing costmap in certain rate?

My PC is equipped with a 6cores 12threads i7 CPU

Best,
Samuel