ros-planning/navigation #456: kinetic-devel does not build on 16.04

mikeferguson, 2016-05-06:
See http://build.ros.org/job/Kdev__navigation__ubuntu_xenial_amd64/1/console

Relevant lines
```
17:20:12 [ 37%] Building CXX object CMakeFiles/costmap_2d.dir/src/costmap_2d_publisher.cpp.o
17:20:12 <command-line>:0:15: warning: missing whitespace after the macro name
17:20:17 [ 40%] Building CXX object CMakeFiles/costmap_2d.dir/src/costmap_math.cpp.o
17:20:17 <command-line>:0:15: warning: missing whitespace after the macro name
17:20:19 [ 42%] Building CXX object CMakeFiles/costmap_2d.dir/src/footprint.cpp.o
17:20:19 <command-line>:0:15: warning: missing whitespace after the macro name
17:20:21 [ 45%] Building CXX object CMakeFiles/costmap_2d.dir/src/costmap_layer.cpp.o
17:20:21 <command-line>:0:15: warning: missing whitespace after the macro name
17:20:26 make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libproj.so', needed by '/tmp/catkin_workspace/devel_isolated/costmap_2d/lib/libcostmap_2d.so'.  Stop.
17:20:26 make[1]: *** [CMakeFiles/costmap_2d.dir/all] Error 2
17:20:26 make: *** [all] Error 2
17:20:26 Command failed, exiting.
17:20:26 CMakeFiles/Makefile2:180: recipe for target 'CMakeFiles/costmap_2d.dir/all' failed
17:20:26 Makefile:138: recipe for target 'all' failed
17:20:26 <== Failed to process package 'costmap_2d': 
17:20:26   Command '['/tmp/catkin_workspace/install_isolated/env.sh', 'make', '-j1']' returned non-zero exit status 2
17:20:26 
```

Appears to be related to VTK6 (by way of PCL), maybe? @tfoote -- I know you guys are blocked by navigation for turtlebot on Kinetic -- have  you seen this before?

tfoote, 2016-05-06:
Yeah, that's odd It is available from libproj-dev http://packages.ubuntu.com/xenial/amd64/libproj-dev/filelist

Some upstream package is not exporting the dependency correctly. 

@jacquelinekay I believe you were dealing with a package that previously included libproj internally and then started using the upstream package.

tfoote, 2016-05-06:
Yeah, I found a commit here: https://github.com/ros/rosdistro/pull/11233 https://github.com/ros-perception/perception_pcl/commit/d4ccc0a2949a26782632125f4c64b25963fb8574 

Re: 
https://github.com/ros-perception/perception_pcl/blob/kinetic-devel/pcl_ros/package.xml#L42-L43

It needs an exec depend on the package too since the package provides a shared lib.

tfoote, 2016-05-06:
Following up a little more it looks like it's actually a bug upstream: https://bugs.launchpad.net/ubuntu/+source/vtk6/+bug/1573234 via https://github.com/ros-perception/perception_pcl/issues/119 

The quickest fix would be to manually add that rosdep as a build depend (and probably a run depend for downstream users convenience) with a comment to remove when upstream is fixed. I'm also going to suggest on the perception_pcl ticket to do the same.

jacquelinekay, 2016-05-06:
there was a conversation about this here: https://github.com/ros-perception/perception_pcl/pull/123#issuecomment-217394576

and a pending pull request to fix it here:

https://github.com/ros-perception/perception_pcl/pull/124/files

mikeferguson, 2016-05-06:
Thanks for the pointers -- I came across some discussion, but google had missed the actual upstream bug. 

It looks like that PR just got opened -- I think the best course of action is to merge that and then release navigation. If for some reason it doesn't get released shortly, I will modify the CMake here and release.

atroyn, 2016-05-08:
I'm seeing this issue even after the PCL changes are merged and local;

```
-- +++ processing catkin package: 'nav_core'
-- ==> add_subdirectory(navigation/nav_core)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at 
/catkin_ws/devel/share/costmap_2d/cmake/costmap_2dConfig.cmake:141 (message):
  Project 'nav_core' tried to find library 'vtkproj4'.  The library is
  neither a target nor built/installed properly.  Did you compile project
  'costmap_2d'? Did you find_package() it before the subdirectory containing
  its code is included?
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  navigation/nav_core/CMakeLists.txt:3 (find_package)
```

mintar, 2016-05-08:
I've commented on the issue over at [perception_pcl#124](https://github.com/ros-perception/perception_pcl/pull/124) .

croesmann, 2016-05-09:
I've tried the same as @atroyn locally with the new perception pre-release from shadow-fixed ([perception_pcl#124](https://github.com/ros-perception/perception_pcl/pull/124)) since navigation is the last blocking dependency of my package and I got the same error. Obviously, adding the *libproj* dependency did not fix the vtk6 issue temporarily. Just wanted to let you know, since I haven't seen a navigation devel job running on the buildfarm with the new perception release yet. By the way, thanks for the explanation @mintar.

closed 2016-05-17 by mikeferguson