ros-planning/navigation #473: [map_server] Improve fall-back behavior for new "mode" field in map.yaml

stwirth, 2016-06-01:
https://github.com/ros-planning/navigation/pull/422 introduced a new variable `mode` for the interpretation of pixel values and removes the previously used `trinary` flag.
If `mode` is not set, it is simply assumed that the mode is `trinary`: https://github.com/ros-planning/navigation/blob/indigo-devel/map_server/src/main.cpp#L128-L130

I want to propose the following improvements:
  1. If the `mode` flag is not set, emit a ROS_WARN that reminds the user to upgrade the map.yaml file to the new format.
  2. If the `mode` flag is not set but a `trinary` field is present in the yaml, set the mode to `TRINARY` if `trinary` is true and to `SCALE` if `trinary` is false.
  3. If both `mode` and `trinary` are present in the yaml, emit a ROS_WARN that the `trinary` field will be ignored (and ignore the `trinary` field).

AchmadFathoni, 2018-05-23:
Oldest open issue

stwirth, 2018-05-30:
I have a workaround in place for my use case, so I don't need this feature. However, I'd be happy to chime in on a PR discussion.

DLu, 2020-07-03:
Solution in #474

DLu, 2020-07-10:
Per discussion in #474, closing this until it becomes a problem again.

closed 2020-07-10 by DLu