ros-planning/navigation #474: [map_server] preserve backwards compatibility of map_server::loadMapFromFile

stwirth, 2016-06-01:
The recent change from `trinary` to `map_mode` in map_server::loadMapFromFile breaks backwards compatibility.
Instead of just changing the signature, a new signature should be provided and the "old" one with the `trinary` boolean should be preserved for backwards compatibility. The users of the old one should be instructed (e.g. by a ROS_WARN) to change their code to use the new interface.

mryellow, 2016-08-16:
Believe this is the offending commit.

https://github.com/ros-planning/navigation/commit/881b49329ba16695065bea7df119c50988a7532d

mryellow, 2016-08-16:
C++ isn't my native language, I imagine that deprecation notice can be done on the compiler side of things and displayed during build instead of as a `ROS_WARN`.

mryellow, 2016-08-16:
Seems it will need a signature without the last arg, which overloads to the new signature. `stdr_simulator` is calling `loadMapFromFile` without that last argument.

```
/home/user/work/ros-kinetic/catkin_ws/src/stdr_simulator/stdr_server/src/map_loader.cpp: In function ‘nav_msgs::OccupancyGrid stdr_server::map_loader::loadMap(const string&)’:
/home/user/work/ros-kinetic/catkin_ws/src/stdr_simulator/stdr_server/src/map_loader.cpp:110:42: error: call of overloaded ‘loadMapFromFile(nav_msgs::GetMap::Response*, const char*, double&, int&, double&, double&, double [3])’ is ambiguous
         res,negate,occ_th,free_th, origin);
```

mryellow, 2016-08-16:
Well that builds, including `world_canvas_editor` and `stdr_server`.

Not sure if I've done it the least verbose way. Imagine adding the third signature without the last arg is how to handle defaults confusing the situation.

mikeferguson, 2017-08-01:
When we originally merged this, I was not treating the "image_loader" library as needing ABI/API compliance because I didn't think anything depended on it. I guess that's not the case, and since we do export it in the CATKIN_LIBRARIES, we should probably preserve the signature.

However, I realize this has sat for almost a year -- and so I have to ask have downstream libraries really not updated already? Is this still relevant given we've broken it for so long?

mryellow, 2017-08-02:
Not sure, haven't been running ROS recently.

Looks like the last build I did used the main repo rather than my fork though.

DLu, 2020-07-03:
@mryellow I apologize for the extreme delays. Is there still downstream code that depends on this? [I have a version with tests](https://github.com/ros-planning/navigation/compare/noetic-devel...DLu:backwards_compatible_map_server) (with the tests from #1006 ) but I don't think we need to merge if there's nothing downstream that still uses the legacy method invocation.

mryellow, 2020-07-03:
All good. No idea, haven't played with ROS for a good while.

DLu, 2020-07-10:
Ok, I'm going to close this for the time being until it becomes an issue again.

closed 2020-07-10 by DLu