ros-planning/navigation #483: move_base doesn't depend on move_base_msgs in its CMakeLists.txt

spmaniato, 2016-06-15:
This is a follow-up to the discussion in http://answers.ros.org/question/236706 and is also related to issues #319 and #373 

In a nutshell, `move_base` declares build and run dependencies on `move_base_msgs` in its [package manifest](https://github.com/ros-planning/navigation/blob/indigo-devel/move_base/package.xml), but there is no mention of `move_base_msgs` in its [CMakeLists.txt](https://github.com/ros-planning/navigation/blob/indigo-devel/move_base/CMakeLists.txt)

Interestingly, this only becomes a problem when building from source with `catkin_make`, i.e., **not** in isolation. I'm guessing that's why it works just fine on the buildfarm. The reason it works with `catkin_make_isolated` is related to the `CPATH` environment variable and is explained in detail in ros/catkin#367 (specifically, in [William's comment](https://github.com/ros/catkin/issues/367#issuecomment-14251186)) Furthermore, `catkin_tools` does something similar (see [workspace mechanics](http://catkin-tools.readthedocs.io/en/latest/mechanics.html)) which is why the problem doesn't show up with `catkin build` either.

If this makes sense to the maintainers I can put together a pull request that patches `CMakeLists.txt`

PS. This should apply to many releases but I've only looked at Indigo and Jade.

mikeferguson, 2016-06-16:
That does seem like an overview -- I would support a PR (please open one against all active branches that have this issue -- likely Indigo, Jade & Kinetic)

spmaniato, 2016-06-16:
Thanks for the quick reply @mikeferguson 😃  OK, I'll put together the patch, test in a fresh VM, and then open the pull request(s)

dcconner, 2016-06-16:
Can you also convert the deprecated _gencpp to _generate_cpp so it will work in Kinetic without warning as well?

spmaniato, 2016-06-22:
I discovered more missing dependencies on messages. The corresponding `_gencpp` targets show up as _warnings_ when building with `clang`:

```
The dependency target "visualization_msgs_gencpp" of target "costmap_2d_markers" does not exist.
The dependency target "geometry_msgs_gencpp" of target "costmap_2d” does not exist.
The dependency target "sensor_msgs_gencpp" of target "costmap_2d_cloud" does not exist.
The dependency target "nav_msgs_gencpp" of target "base_local_planner" does not exist.
The dependency target "nav_msgs_gencpp" of target "trajectory_planner_ros" does not exist.
The dependency target "nav_msgs_gencpp" of target "dwa_local_planner" does not exist.
The dependency target "geometry_msgs_gencpp" of target "move_base" does not exist.
```

Actually, some of these go away when `_gencpp` is replaced with `_generate_messages_cpp`. But a few don't, indicating missing `find_package` components.

I'll try addressing all of them in my pull request(s), including the update to `_generate_messages_cpp`

DLu, 2016-07-27:
Ready to close?

spmaniato, 2016-07-27:
> Ready to close?

Mostly. #504 added `move_base_msgs` as a `find_package()` component, but I think that, technically, we also need an `add_dependencies()` on the corresponding `_generate_messages_cpp `, right?

In addition, `costmap_2d` also seems to be missing from `move_base`'s `find_package()`

I had started implementing and testing _all_ these changes but never got around to cleaning them up and opening a PR 😞 

Having said that, feel free to close this, since the main issue was addressed by #504. I'll eventually put together the larger scope PR 😁

mikeferguson, 2017-07-31:
Addressed in #548

closed 2017-07-31 by mikeferguson

ESDA-LAB, 2018-08-26:
just install `apt-get install libmove-base-msgs-dev`