ros-planning/navigation #487: min_in_place_rotational_vel and min_in_place_vel_theta

willdzeng, 2016-06-25:
In the source code of trajectory_planner_ros.cpp the inplace turning velocity is called

> min_in_place_rotational_vel

but in the ros package wiki it's called 

> min_in_place_vel_theta

spmaniato, 2016-06-25:
Which wiki page(s) specifically?

Here's what I'm seeing (with a shallow search):

* http://wiki.ros.org/base_local_planner ➡️ `min_in_place_vel_theta`
* [base_local_planner / trajectory_planner](https://github.com/ros-planning/navigation/search?utf8=%E2%9C%93&q=min_in_place_vel_theta) ➡️ `min_in_place_vel_theta`
* [base_local_planner / trajectory_planner_ros](https://github.com/ros-planning/navigation/search?utf8=%E2%9C%93&q=min_in_place_rotational_vel) ➡️ `min_in_place_rotational_vel`
 * This line is interesting: 
```cpp
private_nh.param("min_in_place_rotational_vel", min_in_place_vel_th_, 0.4);
```

So I guess there is an inconsistency. Is this what you are referring to?

--

@DLu, did commit 6e0e3e1d07d8abbb730bb1b5fe6a25deea74deb3 from 2013 ever make it to the latest releases? (Also see discussion in #90) Among other things, I think it addresses this very issue:

```cpp
if(private_nh.hasParam("min_in_place_rotational_vel"))
      {
        ROS_WARN("Please use min_in_place_vel_theta instead of min_in_place_rotational_vel");
        private_nh.param("min_in_place_rotational_vel", min_in_place_vel_th_, 0.4);        
      }
      else
      {
        private_nh.param("min_in_place_vel_theta", min_in_place_vel_th_, 0.4);
      }
```

I can't find a branch that contains it:

```shell
$ git branch --contains 6e0e3e1d07d8abbb730bb1b5fe6a25deea74deb3
error: no such commit 6e0e3e1d07d8abbb730bb1b5fe6a25deea74deb3
```

willdzeng, 2016-06-25:
yeah,

http://wiki.ros.org/base_local_planner
wiki page said the parameter's name is 

> min_in_place_vel_theta

but the parameter name is actually

> min_in_place_rotational_vel

I guess the commit you mentioned didn't get in.

Also, even when I increase the min_in_place_rotational_vel, the planner also sometime send small in place turning velocity. Do you have any idea why?>

mikeferguson, 2016-06-28:
Good catch -- a PR (with the backwards compatibility mentioned in the DLu commit above) could definitely be accepted into all of indigo/jade/kinetic if someone wants to go about opening them.

mikeferguson, 2018-05-22:
Is in PR -- closing so I can keep track of what we have to actually still do before release!

commit dc44a830 (2018-05-22): fix param to min_in_place_vel_theta, closes #487
commit 2dfa08f2 (2018-05-22): fix param to min_in_place_vel_theta, closes #487
commit 611e8cc6 (2018-05-29): fix param to min_in_place_vel_theta, closes #487
closed 2018-05-22 by mikeferguson