ros-planning/navigation #488: Local Costmap Padding?

willdzeng, 2016-06-25:
[image]
[image]

Because I found out the padding of local costmap doesn't work.
I try to visualize the obstacle cost cloud, and found out the padding of costmap and the padding of obstacle cloud is not the same. 
I am wondering if this is a bug, or it's just the implementation?

spmaniato, 2016-06-26:
@willdzeng Which distro and which version (release) are you running? I think what you're seeing might be related to #485, which was fixed for Indigo (in #486) two days ago.

willdzeng, 2016-06-26:
I am using the latest jade-devel, I think that commit is not merged into jade

willdzeng, 2016-06-26:
Also I found out the local costmap didn't update after the obstacle are removed even after I used the coomit in #486 . 
 For example, in the following screenshots.
The point cloud showed that the obstacle is removed, but the local costmap didn't update.
[image]
[image]

willdzeng, 2016-06-26:
Inflation still doesn't show in the obstacle cost point cloud visualization. This is using the latest indigo-devel branch 

[image]
[image]

mikeferguson, 2016-06-28:
I think you are seeing several different things here:

 * On the initial post about the point_cloud vs the costmap -- I'm pretty sure that if the costmap is up to date, it will be properly taken into account in the planner -- for instance, the robot radius appears to be all "very high cost". However, depending on the settings you have for the planner costs, the actual cost may drop off quite steeply in the lower cost inflation region (this is pretty commonly done to avoid robots having difficult times in narrow aisles) -- I'm going to close this part of the ticket because I don't think it is actually a bug without further information.

 * On the "not clearing" -- you just don't have any data to clear with -- it looks like you are using something like a primesense sensor, and the sensor has very few returned data points in the center region of the camera, probably because the sensor can't see far enough to actually see anything. Lack of data does not cause clearing -- you need actual clearing endpoints to raytrace and actually clear out the previous observations. With short-range sensors this can be quite difficult -- and I've seen many people actually create a second point cloud where all nans/infs/etc are made into "max_range-epsilon" so that lack of data actually does clear -- however, it should be noted that this is not without its own caveats. In the future, this sort of "how do I do X?" really belongs on ROS answers, not in a ticket.

closed 2016-06-28 by mikeferguson

willdzeng, 2016-06-28:
OK~, I got you. Sorry, I thought it was a bug of the map_viz or inflation didn't take into account. I will try change the cost_scaling_factor to see if it changes.

mikeferguson, 2016-06-28:
No problem -- especially on that map_viz part, I could see that looking like a bug at first.

One other thought -- you can probably change the scale of min/max in your RVIZ settings, there is probably a very shallow slope across the red region, but the scaling is probably so small compared to the other color differences that it all shows up red.