ros-planning/navigation #492: Compilation error

sylvainar, 2016-06-30:
Hey guys !

I'm trying to compile using catkin_make, but I've got an error : 

`/home/summit/catkin_ws_sylvain/src/navigation/clear_costmap_recovery/src/clear_costmap_recovery.cpp: In member function 'void clear_costmap_recovery::ClearCostmapRecovery::clearMap(boost::shared_ptr<costmap_2d::CostmapLayer>, double, double)':
/home/summit/catkin_ws_sylvain/src/navigation/clear_costmap_recovery/src/clear_costmap_recovery.cpp:127:22: error: 'mutex_t' is not a member of 'costmap_2d::Costmap2D'
`
And after, lots more errors. It seems that something missing but I don't know what and why, I've cloned the repository and added dependencies..

Thanks !

DLu, 2016-07-01:
What version of the repository are you running, and on what system? 

Also, until this is confirmed as a bug, its probably better to tackle this on answers.ros.org.

mikeferguson, 2017-07-31:
No response, closing

closed 2017-07-31 by mikeferguson