ros-planning/navigation #494: cmd_vel flooded with zero-velocity commands during recovery behaviors

corot, 2016-07-05:
This is something new, seems to me that coming with [this commit](https://github.com/ros-planning/navigation/commit/0a686c90c6f188a2731f6f88c2c08610f0ec907e). Happens when the global planner fails to produce a plan: move_base keeps publishing zero-velocity messages. This can hamper recovery behaviors that want to move the robot.
@gjc13, do you think we can remove the publishZeroVelocity(); part from your fix?

mikeferguson, 2016-07-06:
I hadn't thought of that sort of side effect when reviewing/merging that -- @corot if you open a PR with a revert to that part (against i/j/k branches) we can get that merged and released shortly

corot, 2016-07-06:
Sure,  i'll do later today (CET). I'll remove just the publishZeroVelocity();, as I have no idea what the other change in the commit does. Would be great if @gjc13 can seed some light on that.

gjc13, 2016-07-07:
In our configuration we disabled the recovery behaviour, so this part is not tested. Sorry for the side effect. I think you can remove it.

corot, 2016-08-24:
Fixed; thanks! Can be closed.
Seems like there's no problem on J/K. I didn't tried the code, but the line I removed on indigo is not there

commit 37c515b8 (2016-08-15): Merge pull request #495 from corot/patch-3

Fix #494: prevent zero-velocity commands during recovery behaviors
commit 8d2001be (2016-08-15): Fix #494: prevent zero-velocity commands during recovery behaviors

Partially reverts 0a686c90c6f188a2731f6f88c2c08610f0ec907e
closed 2016-08-24 by corot