ros-planning/navigation #496: Can we make move_base fail faster if the global planner cannot find a path?

corot, 2016-07-11:
Now move_base keeps calling global planner up to 'planner_patience' seconds regardless how many times the global planner fails. In our particular case, we allocate 2 seconds (rather short time) to the SBPL lattice planner, so we set planner_patience as 2.5 seconds. This is a notable waste of time because the planner has time to fail many many times, while after 4 or five attempts planner can do in some tens of milliseconds, it seems clear he will never succeed.

Afaik, there's no solution for this without patching move_base. Would it make sense to replace planner patience by an alternative planner_max_calls parameter? Or better keep both mechanism for different use cases.

mikeferguson, 2016-07-11:
removing planner_patience would probably be too drastic of an API change -- but adding an additional "max_planning_attempts" is an acceptable API change as long as the default is "disabled" (probably represented as either 0 or -1)

corot, 2016-07-11:
Thanks! Also acceptable to release on indigo?

mikeferguson, 2016-07-11:
Yes -- we'll have to extensively review it (so the shorter the PR, the better), but we should be able to put this into I/J/K as long as we maintain backwards-comparability

corot, 2016-07-12:
I tried to be quirurgical...

closed 2016-08-15 by mikeferguson