ros-planning/navigation #497: costmap_2d documentation does not inform it subscribe /map topic

tibloop, 2016-07-12:
The documentation does not inform move_base subscribe /map topic through costmap_2d and static_layer . Also it should inform we can use the parameter map_topic to set it.

spmaniato, 2016-07-13:
It does here: http://wiki.ros.org/move_base#move_base-1 (figure) and here: http://wiki.ros.org/costmap_2d/hydro/staticmap (both topic and parameter) That's because only the `static_layer` (i.e., usually the global costmap) needs the map.

You can see the ROS API of all layers here: http://wiki.ros.org/costmap_2d?distro=kinetic#costmap_2d.2BAC8-layered.Layer_Specifications

tibloop, 2016-07-14:
yes, but if you don't know the package, its quite a challenge to discover that you need to open costmap_2d documentation and then static_layer documentation to get this information. I think move_base documentation should at least inform that costmap_2d is the node used for global_costmap. Maybe in the figure. I realise the information are here, just not accessible to persons unfamiliar with the package.. Its just a proposition, you can ignore it ^^ I am just trying to help people not spending hours trying to find how to change the topic name like I did ! :)

spmaniato, 2016-07-14:
OK. I think what you were trying to accomplish is easier using topic remapping:
```
<remap from="map" to="$(arg map_topic)" />
```
See, for example, these launch files: [fetch_navigation/launch](https://github.com/fetchrobotics/fetch_ros/tree/indigo-devel/fetch_navigation/launch) (The usecase here is a "keepout" map.)

PS. For the record, I'm not a package maintainer. Just a user/developer trying to be helpful 😛

mikeferguson, 2016-07-14:
Section 1.1.1 states that move_base is what ties all the different components together, and section 1.1.7 of the move_base wiki page states what components exist and links to costmap_2d. The reality here is that move_base is quite complex -- you have to really read the whole documentation to understand what is going on, and there is just no getting around that.

On the costmap_2d page, the plugins are pretty poorly described, more like a footnote at the end of the page, but the parameters are described in the linked pages.

Improvements are certainly welcome -- anyone can create an account and improve/modify the wiki. As such, I'm going to close this ticket since there isn't a specific course of action that can be verifiably closed, and any updates to the wiki do not actually require changes in this repository.

closed 2016-07-14 by mikeferguson