ros-planning/navigation #499: Inconsistent odom_alpha1 and odom_alpha4 parameters between AMCL's diff and omni models?

spmaniato, 2016-07-13:
I've been working on AMCL lately and noticed the following inconsistency in [`amcl_odom.cpp`](https://github.com/ros-planning/navigation/blob/indigo-devel/amcl/src/amcl/sensors/amcl_odom.cpp)

Let's start with the documentation. The [amcl wiki page](http://wiki.ros.org/amcl) says (abbreviated):

 - `~odom_alpha1` Expected noise in odometry's rotation estimate from the rotational component
 - `~odom_alpha2` Expected noise in odometry's rotation estimate from translational component
 - `~odom_alpha3` Expected noise in odometry's translation estimate from the translational component
 - `~odom_alpha4` Expected noise in odometry's translation estimate from the rotational component

In the source code ([amcl_odom.cpp](https://github.com/ros-planning/navigation/blob/indigo-devel/amcl/src/amcl/sensors/amcl_odom.cpp)), we see the following for the `diff` and `omni` models:

- For `diff`:

 ```cpp
    delta_rot1_hat = angle_diff(delta_rot1,
              pf_ran_gaussian(this->alpha1*delta_rot1_noise*delta_rot1_noise +
                              this->alpha2*delta_trans*delta_trans));
    delta_trans_hat = delta_trans - 
              pf_ran_gaussian(this->alpha3*delta_trans*delta_trans +
                              this->alpha4*delta_rot1_noise*delta_rot1_noise +
                              this->alpha4*delta_rot2_noise*delta_rot2_noise);
    delta_rot2_hat = angle_diff(delta_rot2,
              pf_ran_gaussian(this->alpha1*delta_rot2_noise*delta_rot2_noise +
                              this->alpha2*delta_trans*delta_trans));
 ```
So the `diff` model's implementation is consistent with the wiki page's explanation.

- However, for `omni`:

 ```cpp
    double trans_hat_stddev = (alpha3 * (delta_trans*delta_trans) +
                               alpha1 * (delta_rot*delta_rot)); // <-- Look here,
    double rot_hat_stddev = (alpha4 * (delta_rot*delta_rot) + // <-- here,
                             alpha2 * (delta_trans*delta_trans));
    double strafe_hat_stddev = (alpha1 * (delta_rot*delta_rot) + // <-- and here.
                                alpha5 * (delta_trans*delta_trans));
 ```

This seems to imply that, for the `omni` model,:

 - `alpha1` is **translation** noise from rotation, not rotation from rotation
 - `alpha4` is **rotation** noise from rotation, not translation from rotation

In other words, `alpha1` and `alpha4` look flipped in the `omni` model (compared to `diff` and the wiki page). The other two common params, `alpha2` and `alpha3`, are OK. The inconsistency also appears in the `corrected` versions of the two odom models btw.

What do you think? Seems inconsistent to me.

If I'm right, I'd be down for submitting a pull request. But I'm not sure against which distro, as a proper fix would break people's I/J/K AMCL configuration.

**PS1**. I've "verified" this in simulation by setting one of the inconsistent noise params to a large number and seeing how the particles spread when I send various velocity commands.
**PS2**. Originally posted on ROS Answers: http://answers.ros.org/question/239245

mikeferguson, 2016-07-14:
That does appear to be flipped -- it's way too late to fix this in I/J -- maybe acceptable in K, but questionable given that K has been "released" for about 2 months now. It may actually be better to just update the documentation and note the difference for omni.

spmaniato, 2016-07-14:
Agreed. Can we label this issue and leave it open for a future pull request against Lunar, once that branch is up?

I'll comment here again with a link to the modified wiki page section.

mikeferguson, 2018-03-28:
Targeting Melodic

mikeferguson, 2018-05-22:
Is in PR -- closing so I can keep track of what we have to actually still do before release!

closed 2018-05-22 by mikeferguson

EwingKang, 2019-06-26:
Apparently, the Kinetic version is still using version 1.14.4. (Melodic 1.16.2)
http://repositories.ros.org/status_page/ros_kinetic_default.html?q=amcl
However this AMCL fix happened somewhere between 1.14.4 and 1.16.2.
The wiki page for kinetic only shows the new version. http://wiki.ros.org/amcl 
Not sure how should this be fixed?

EwingKang, 2019-06-27:
I've edited [the wiki page](http://wiki.ros.org/amcl?distro=lunar) so there won't be others that is trying to tune the AMCL spent many hours finding problems like me.

commit f1e0e659 (2018-05-22): fix swapped odom1/4 in omni model, fixes #499
commit 31c8e948 (2018-05-29): fix swapped odom1/4 in omni model, fixes #499