ros-planning/navigation #503: When move_base goal is set in a certain direction, dwa_local_planner plans less stably and the robot is moving slower

zkytony, 2016-07-23:
I found some interesting behavior in dwa_local_planner, not sure if it is a bug, or if it can be resolved by setting parameters differently.

The behavior is that, when the goal is set in the -x direction with respect to TF origin, dwa local planner plans less stably (the local planned trajectory jumps around) and the robot moves really slowly. But when the goal is set in the +x direction, dwa local planner is much more stable, and the robot can move faster.

I demonstrated this inconsistency in the following two videos (on youtube):
* [empty environment](https://www.youtube.com/watch?v=Tl3uU39Kj54&feature=youtu.be):
* [environment with obstacles](https://www.youtube.com/watch?v=rV3IRxkxTNk&feature=youtu.be)
In these videos, the local trajectory planned by dwa local planner is colored blue. The red line is the global path used by dwa local planner. The green line (in the second video) is the global path produced by global_planner. The dwa parameters I used are:

## DWAPlannerROS:

## Robot Configuration Parameters
  max_trans_vel: 0.7
  min_trans_vel: 0.1

  max_vel_x: 0.7
  min_vel_x: -0.05

  max_rot_vel: 1.2
  min_rot_vel: 0.2

  acc_lim_x: 1.0
  acc_lim_theta: 2.0
  acc_limit_trans: 2.0

## Differential-drive robot configuration
  holonomic_robot: false
  acc_lim_y: 0.0
  min_vel_y: 0.0
  max_vel_y: 0.0
  vy_samples: 0

## Goal Tolerance Parameters
  yaw_goal_tolerance: 0.2
  xy_goal_tolerance: 0.2
  latch_xy_goal_tolerance: true

## Minimum velocity for the robot to be considered not stop
  trans_stopped_vel: 0.1
  rot_stopped_vel: 0.1

## Forward Simulation Parameters
  sim_time: 2.0
  vx_samples: 20
  vy_samples: 0
  vth_samples: 40
  penalize_negative_x: false #no longer used

## Trajectory Scoring Parameters
  path_distance_bias: 32.0
  goal_distance_bias: 9.0
  occdist_scale: 0.020
  stop_time_buffer: 0.5

# Oscillation Prevention Parameters
  oscillation_reset_dist: 0.05

# Global Plan Parameters
  prune_plan: true

Can anyone reproduce such behavior? If so, should we consider this a bug?

If any clarification is helpful, please point out.

iretiayo, 2017-06-23:
We're having the same issue but in the negative direction. Navigation works well when the goal  is set in the -x direction with respect to the map origin, but it doesn't work when the goal is in the +x direction. Is there a fix for this yet?

mikeferguson, 2017-08-01:
I've never seen this. One thing that does jump out to me about your config -- why is min_vel_x negative? For most people this is slightly positive (most robots can't actually move at infinitesimally small velocities), or at least 0.

zkytony, 2017-08-02:
`min_vel_x` was negative because we want the robot to be able to backup.

mikeferguson, 2017-08-02:
Out of curiosity, if you set it to slightly positive, does this slowness still occur?

zkytony, 2017-08-17:
@mikeferguson Good question. I haven't tested that. I'll test it soon. If that changes anything, the move_base local planner would be even more strange...because in those videos, the robot only needs to move forward. I'm not sure what role the minimum velocity plays in the empty environment video, in particular.

tcassell94, 2017-08-30:
Don't know if you've had any resolution to this, but your config for the x and trans velocities seems off. If your robot is non-holonomic, the values for trans and x velocities/accelerations should match exactly, since the y velocities are always zero. The exception is the min_trans_vel which should be the same magnitude as your min_vel_x, but your min_vel_x may be negative if you want to drive backwards. The mismatches you have may be causing erratic planning, but I wouldn't have high hopes that it will it solve your issue.

zkytony, 2018-05-28:
I have not worked on navigation for a while. I'm closing this issue. If anyone has an update they can reopen it.

closed 2018-05-28 by zkytony

meiqua, 2018-08-03:
I have a guess. Obviously -x and +x direction should be the same, so there must be somewhere break the symmetric. Due to the DWA trajectory sampling strategy, velocities are always evenly sampled from min to max. So when time is not enough for scoring trajectory, -x direction only explore velocity close to min, while +x direction can explore velocity close to max.
If my guess is true, a simple solution is to improve sampling in a coarse-to-fine stepping manner.

davidteixeira, 2019-09-30:
The same thing happens to me. It seems related to the odometry frame. So if the stack is launched with the robot facing North, all goals in the North direction cause the robot to move quickly (achieving max speed if there are no obstacles), while all goals in the South direction cause the robot to move very slowly. In my case, max_vel_x = 1.0, achievable in the "fast direction". In the "slow direction", the maximum velocity is around 0.2 m/s. Anyone have any info on this? I have been digging around the trajectory generator code, but no luck yet...