ros-planning/navigation #507: Message in dwa_planner_ros should use debug type instead of info

pitschm1, 2016-07-27:
In the dwa_planner_ros.cpp file in all of the devel branches, there is a ROS_INFO statement that prints "Got new plan" every time the plan changes a little. This can quickly fill up the terminal while trying to reach a specific goal point if info statements are shown for move_base. It is not an error, but it can get annoying and affect the user seeing useful information. 

Should this instead be a ROS_DEBUG so that the user can control it with verbosity? It does not provide information that is informative for common use.

PeterMitrano, 2016-08-01:
Sometime that's incredibly useful--I like having it printed. I don't think there's enough reason to change it. There are *tons* of ROS_DEBUG statements that I don't want turned on along with it. My guess is this won't get enough support.

QinZiwen, 2016-12-03:
My terminal show that DWA planner failes to produce path. What are there possible reason? 
Thank you very much!

mikeferguson, 2017-07-28:
Seems there isn't a strong reaction to implementing this, going to close.

closed 2017-07-28 by mikeferguson