ros-planning/navigation #508: footprint_clearing_enabled param: not all footprint area is cleared

corot, 2016-07-29:
Seems to me that cells with center laying outside the footprint are still marked as obstacles (not cleaned), what probably makes sense. But if a corner of the cell is within the footprint, move_base always fails.

[image]

Shouldn't we clean ALL cells with any portion within footprint area?

I cannot remember if that was the case before removing footprint_layer (issue #289)

panshengjie, 2016-07-29:
hello   
I  just fixed it my self by turn off  footprint legal check in TrajectoryPlanner::generateTrajectory
try to  replace  code :  double footprint_cost = footprintCost(x_i, y_i, theta_i);  with code: double footprint_cost=0; 
that works great but the robot almost wont escape anymore
but still can avoid obstacles

PeterMitrano, 2016-08-01:
This is probably a side effect of using bresenham's line tracing algorithm
[https://en.wikipedia.org/wiki/Bresenham%27s_line_algorithm](https://en.wikipedia.org/wiki/Bresenham%27s_line_algorithm)

I think there's a line_iterator class somewhere that implements it. Although TBH I can't actually understand your picture. Which part is insid ethe robot and why are there lines of different colors?

corot, 2016-08-08:
Sorry @PeterMitrano, the picture is not a very fortunate example. Robot center is to the right-up of the green line. The black line is part of the RViz grid that I should have disabled.

PeterMitrano, 2016-08-08:
ok so is it the red dot you're talking about? or just all the red/blue stuff to the right-up of the green line

corot, 2016-08-08:
[image]

Hope this image is clearer: as you can see, the obstacles (yellow cells) overlap the robot footprint (green). So move_base_will always fail.

PeterMitrano, 2016-08-08:
Yep, that's probably because of the line tracing algorithm. Look at the image in the wiki article I sent you. The obstacle layer (probably same for voxel) uses methods that eventually boil down to `costmap_2d::Costmap2D::polygonOutlineCells`, which uses `costmap_2d::Costmap2D::raytraceLine`. I don't think we should modify the ray trace algorithm. Instead, maybe just add one cell on all sides when we're filling in the polygon? Not sure how the maintainers feel about any of this. Look at the code and maybe we can agree on something and make a PR.

corot, 2016-08-18:
Current implementation with Bresenham's line algorithm looks just right... and I don't feel confident enough to propose an alternative, so... unless more people have the same problem, I'll close this issue by now, and live with it (I created a recovery behavior that shrinks the footprint)

closed 2016-08-18 by corot

PeterMitrano, 2016-08-18:
haha oh my god that's a fantastic recovery behavior

corot, 2016-08-18:
hehe...  risky, but works many times!  :fearful:

sinhau, 2016-12-09:
@corot Would be great if you could share your recovery behavior code.  Sounds like a really interesting idea!