ros-planning/navigation #516: TrajectoryPlanner do not consider obstacles while generating escape velocities.

AravindaDP, 2016-08-24:
It appears that when TrajectoryPlanner generates escape trajectories, It do not consider the costmap obstacles.

https://github.com/ros-planning/navigation/blob/indigo-devel/base_local_planner/src/trajectory_planner.cpp#L877

Is this by design? I think this behavior need to be at least made configurable so that users could choose to whether or not to consider costmap obstacles when escaping.

AravindaDP, 2016-09-26:
This is also discussed at http://answers.ros.org/question/217043/escape-velocity-robot-behavior-move_base/

AravindaDP, 2016-10-31:
@DLu , @mikeferguson 
Thought you might be able to shed some light on this?

mikeferguson, 2017-08-01:
I think the notion was: most robots had a single forward-facing laser and didn't have sensors behind them, and so they probably have an inaccurate understanding of the world back there so checking the costmap was already fairly useless.

That's probably no longer the case in many cases, and so adding (optional) costmap checking sounds like a great enhancement. Pull requests welcomed.

Cavalletta98, 2023-02-13:
Has been solved?

mikeferguson, 2023-02-13:
AFAIK, nobody has worked on this (it certainly hasn't been proposed as a PR to this repo even if they have)