ros-planning/navigation #518: amcl segfault in libamcl_map.so

nxdefiant, 2016-08-31:
amcl segfaults when compiled from source in current debian testing (gcc 6.1.1, boost 1.61, kinetic-devel)

ros debug output is attached in amcl.txt:

[amcl.txt](https://github.com/ros-planning/navigation/files/448150/amcl.txt)

nxdefiant, 2016-09-04:
The crash is in map/map_cspace.cpp, enqueue():

```
unsigned int di = abs(i - src_i);
unsigned int dj = abs(j - src_j);
```
(i, j, src_i, src_j also unsigned int)

with e.g. i=997 and src_i=998 on g+ 6.1.1 it results in 4294967295 instead of 1 because g++ now has its own abs() in math.h. The values should probably cast to signed or the lines converted to something like

`unsigned int di = i > src_i ? i - src_i : src_i - i;`

mikeferguson, 2017-07-28:
Likely fixed by #587

closed 2017-07-28 by mikeferguson