ros-planning/navigation #523: Global planner called much more times than expected

corot, 2016-09-29:
After merging #498, if max_planning_retries is set to 0 or bigger, global planner should be called 1 + max_planning_retries times. If not, it should be call for up to planner_patiente seconds.
But it's not the case; it gets called tens or even hundreds of times!

This is discussed in this [ROS answer](http://answers.ros.org/question/242629/global-planner-behavior-upon-planning-failure/)

asimay, 2017-02-05:
+1.
I'm writing another global planner demo now, and find this issue also. Original global planner is called back much more times than we expected, even though we send relative "goal" msg only once.

asimay, 2017-02-05:
I can test this in these days. will give you feedback later. Thanks!

asimay, 2017-02-09:
hi, dear @corot , I test the new changes.
And it works well in normal case.
but if the robot stuck in recovery mode, the "makePlan" still works too much time.  

Backgroud: if the goal set too near the inflation layer or in inflation layer, this is made deliberated, the robot will stuck in recovery mode, and oscillate over and over again, and this made the max_retries invalid.

please see screenshot as belows: (green line is the plan path, we can see at least 3 lines here)

[image]

and short log files as belows:

> [ INFO] [1486616322.730474749, 11.900000000]: Got new plan
[ INFO] [1486616322.763460791, 11.900000000]: Got new plan
[ WARN] [1486616397.623080855, 86.800000000]: DWA planner failed to produce path.
[ INFO] [1486616397.635274320, 86.800000000]: Got new plan
[ WARN] [1486616406.830471817, 96.000000000]: Clearing costmap to unstuck robot (3.000000m).
[ INFO] [1486616406.845336966, 96.000000000]: Got new plan
[ WARN] [1486616416.930792445, 106.100000000]: Rotate recovery behavior started.
[ INFO] [1486616419.399480928, 108.600000000]: Got new plan
[ WARN] [1486616429.511433472, 118.700000000]: Clearing costmap to unstuck robot (1.840000m).
[ INFO] [1486616429.530771139, 118.700000000]: Got new plan
[ WARN] [1486616439.629621073, 128.800000000]: Rotate recovery behavior started.
[ INFO] [1486616445.708081987, 134.900000000]:[ERROR] [1486616455.830933941, 145.000000000]: Aborting because the robot appears to be oscillating over and over. Even after executing all recovery behaviors

Thank you!

corot, 2017-02-09:
Hi @asimay , thanks a lot for testing the PR!
which value you have in max_planning_retries? Then, can you count the calls to makePlan? if the PR is correct, I suppose it should get called (max_planning_retries + 1) * <number of recovery behaviors>. If not the case, you are right, there's something wrong in the PR.

asimay, 2017-02-11:
Hi, Dear @corot , I set `max_planning_retries=0` . the calls maybe 6-7 times in my case.
yes, in normal case, it works good if the goal is correct and turtlebot works good, the called time is  exactly (max_planning_retries + 1).
it is only uncertain in abnormal cases, for example, the goal coordinates is in obstacle's inflation_radius or too close to obstacle, this will caused the turtlebot local plan failed and made turtlebot oscillated nearby the goal, and finally made the turtlebot run into recovery mode and cause the re-makePlan case occur.
,

mikeferguson, 2017-08-01:
Fix should be merged shortly.

closed 2017-08-01 by mikeferguson