ros-planning/navigation #537: Indigo build fails when ROS java is installed

jacobperron, 2016-11-02:
Several packages fail to build. For example, **navfn** fails:

```bash
Traceback (most recent call last):
  File "/opt/ros/indigo/share/genjava/cmake/../../../lib/genjava/genjava_gradle_project.py", line 14, in <module>
    genjava.main(sys.argv)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genjava/genjava_main.py", line 82, in main
    gradle_project.create(args.package, args.output_dir)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genjava/gradle_project.py", line 152, in create
    raise IOError("could not find %s on the ros package path" % msg_pkg_name)
IOError: could not find navfn on the ros package path
```

and **robot\_pose\_ekf**:

```bash
Traceback (most recent call last):
  File "/opt/ros/indigo/share/genjava/cmake/../../../lib/genjava/genjava_gradle_project.py", line 14, in <module>
    genjava.main(sys.argv)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genjava/genjava_main.py", line 82, in main
    gradle_project.create(args.package, args.output_dir)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genjava/gradle_project.py", line 152, in create
    raise IOError("could not find %s on the ros package path" % msg_pkg_name)
IOError: could not find robot_pose_ekf on the ros package path
```

But the build succeeds upon removal of ROS java ([Ref](http://answers.ros.org/question/213859/failed-to-build-razor_imu_9dof/?answer=222562#post-id-222562)) with the following command:

```bash
sudo apt-get remove ros-indigo-genjava
```

mikeferguson, 2016-11-02:
This would appear to actually be an issue with either rosjava or your environment/paths. I doubt there is anything we can do within the navigation stack to fix this, so I would suggest following up with the rosjava devs.

closed 2016-11-02 by mikeferguson

mintar, 2016-12-01:
Actually, this is an issue caused by incorrect dependencies in the navigation stack. I've fixed it in #548 . :)

caicaixia, 2017-01-10:
$ sudo apt-get remove ros-indigo-genjava

caicaixia, 2017-01-10:
it works if your removed ros-indigo-genjava.  I think it maybe the version of qt5, and qt4

mintar, 2017-01-10:
Nope, this has nothing to do with qt. The fix for this issue is #548. Removing genjava is not an option if you need rosjava.

caicaixia, 2017-01-10:
Hello,  I don't use the navigation stack, but also have this error.
So what can I do?  What is the solution?  Did you already merge the changes to the ros indigo?

caicaixia, 2017-01-10:
Should I update the ros-indigo-navigation?

mintar, 2017-01-11:
> Hello, I don't use the navigation stack, but also have this error.

The bug only occurs when rosjava is installed *and* navigation is checked out as a source package into the Catkin workspace you're building. The binary package `ros-indigo-navigation` has nothing to do with this. If you don't have navigation checked out from source and you still have some error message, it's a different bug. Please ask on http://answers.ros.org .

PR #548 (2017-07-31): Fix CMakeLists + package.xmls
This fixes compilation errors when rosjava is installed on the system.
It also removes a lot of errors reported by catkin_lint.

Fixes #537 .
commit d443c448 (2017-07-31): Fix CMakeLists + package.xmls (#548)

* Fix CMakeLists + package.xmls

This fixes compilation errors when rosjava is installed on the system.
It also removes a lot of errors reported by catkin_lint.

Fixes #537 .

* Fix more CMakeLists.txt + package.xmls

* amcl: disambiguate include path

This fixes the following catkin_lint warning:

    amcl: warning: include paths 'include/amcl/map' and 'include' are ambiguous
         * You have used two include paths where one is a parent of the
         * other. Thus the same headers can be included with two different
         * include paths which may  [... cut]
commit a013a01a (2017-07-31): Fix CMakeLists + package.xmls (#548)

* Fix CMakeLists + package.xmls

This fixes compilation errors when rosjava is installed on the system.
It also removes a lot of errors reported by catkin_lint.

Fixes #537 .

* Fix more CMakeLists.txt + package.xmls

* amcl: disambiguate include path

This fixes the following catkin_lint warning:

    amcl: warning: include paths 'include/amcl/map' and 'include' are ambiguous
         * You have used two include paths where one is a parent of the
         * other. Thus the same headers can be included with two different
         * include paths which may  [... cut]
commit 2c4d5503 (2017-07-31): Fix CMakeLists + package.xmls (#548)

* Fix CMakeLists + package.xmls

This fixes compilation errors when rosjava is installed on the system.
It also removes a lot of errors reported by catkin_lint.

Fixes #537 .

* Fix more CMakeLists.txt + package.xmls

* amcl: disambiguate include path

This fixes the following catkin_lint warning:

    amcl: warning: include paths 'include/amcl/map' and 'include' are ambiguous
         * You have used two include paths where one is a parent of the
         * other. Thus the same headers can be included with two different
         * include paths which may  [... cut]