ros-planning/navigation #540: switch to Kinetic

jspricke, 2016-11-12:
I would propose to switch the default branch on Github to kinetic-devel and merge the changes from jade-devel.

mikeferguson, 2016-11-13:
Switched to kinetic-devel.

I'm not sure I understand the comment about merging from jade-devel, we forked because there are API/ABI differences between jade/kinetic, so we still need the jade-devel branch, and AFAIK any new features put into jade-devel are also in kinetic-devel (unless we accidentally missed something, and if so, we should forward-port the individual missed PRs).

jspricke, 2016-11-13:
* Michael Ferguson <notifications@github.com> [2016-11-13 13:46]:
> Switched to kinetic-devel.

Thanks!

> I'm not sure I understand the comment about merging from jade-devel, we forked because there are API/ABI differences between jade/kinetic, so we still need the jade-devel branch, and AFAIK any new [...]

Then it's fine, I just noticed a lot of pull requests against old
branches.

closed 2016-11-13 by mikeferguson