ros-planning/navigation #542: amcl odom_alpha parameters have wrong description in wiki.ros.org

luke321, 2016-11-17:
It took me quite some time till I figured out that something is wrong with the documentation on wiki.ros.org, I hope this gets fixed to spare others some time.

The Description texts actual describe this values:
alpha4
alpha2
alpha3
alpha1
alpha5

alpha1 and alpha4 are switched

CURRENT Description in wiki.ros.org:

> ~odom_alpha1 (double, default: 0.2) 
Specifies the expected noise in odometry's rotation estimate from the rotational component of the robot's motion. 
~odom_alpha2 (double, default: 0.2) 
Specifies the expected noise in odometry's rotation estimate from translational component of the robot's motion. 
~odom_alpha3 (double, default: 0.2) 
Specifies the expected noise in odometry's translation estimate from the translational component of the robot's motion. 
~odom_alpha4 (double, default: 0.2) 
Specifies the expected noise in odometry's translation estimate from the rotational component of the robot's motion. 
~odom_alpha5 (double, default: 0.2) 
Translation-related noise parameter (only used if model is "omni"). 

actual code in amcl_odom.cpp:

`double trans_hat_stddev = sqrt( alpha3 * (delta_trans*delta_trans) +
                                    alpha1 * (delta_rot*delta_rot) );
    double rot_hat_stddev = sqrt( alpha4 * (delta_rot*delta_rot) +
                                  alpha2 * (delta_trans*delta_trans) );
    double strafe_hat_stddev = sqrt( alpha1 * (delta_rot*delta_rot) +
                                     alpha5 * (delta_trans*delta_trans) );`

This would be the CORRECT Description:

> ~odom_alpha1 (double, default: 0.2) 
Specifies the expected noise in odometry's translation estimate from the rotational component of the robot's motion. 
~odom_alpha2 (double, default: 0.2) 
Specifies the expected noise in odometry's rotation estimate from translational component of the robot's motion. 
~odom_alpha3 (double, default: 0.2) 
Specifies the expected noise in odometry's translation estimate from the translational component of the robot's motion. 
~odom_alpha4 (double, default: 0.2) 
Specifies the expected noise in odometry's rotation estimate from the rotational component of the robot's motion. 
~odom_alpha5 (double, default: 0.2) 
Translation-related noise parameter (only used if model is "omni").

DLu, 2016-11-17:
Were you not able to make the change on the wiki?

spmaniato, 2016-11-18:
@luke321 @DLu It's a bit more nuanced than that: #499 

In a nutshell, `alpha1` and `alpha4` look flipped in the `omni` model(s) source code (compared to `diff` and the wiki page)