ros-planning/navigation #572: Lunar release

mikaelarguedas, 2017-04-25:
Hi @mikeferguson, @DLu and navigation stack maintainers!

As you may know the next ROS release Lunar Loggerhead is around the corner :tada: 
While there is still a dependency not satisfied in order to release the nav stack (pcl_perception should land in the next day or two), I'm poking you now so that you can get started on the work needed to get this released. 

While not being required by desktop-full, it would be really awesome to have the nav stack as part of the initial Lunar release.

mikeferguson, 2017-04-25:
Release should be pretty straight-forward once the depends land -- especially since someone already released BFL (my most hated ROS package). If you would be so kind as to poke again, I can probably release over the weekend.

mikaelarguedas, 2017-04-25:
Great news! 
>  If you would be so kind as to poke again once that depend has cleared, I can probably release over the weekend.

will do, 

thanks for the quick reply, appreciate it

mikaelarguedas, 2017-04-26:
@mikeferguson FYI: All dependencies [have been released](http://repositories.ros.org/status_page/blocked_releases_lunar.html?q=id%3D%22navigation%22&s=6&r=1) and built successfully :tada:

mikaelarguedas, 2017-05-08:
@mikeferguson Friendly ping

mikaelarguedas, 2017-05-20:
@mikeferguson FYI: I ran the prerelease tests and it looks like navigation can be released into lunar with very few changes. The only error I faced is described in and fixed by https://github.com/ros-planning/navigation/pull/578

DLu, 2017-05-30:
There is now a lunar branch. (Just lunar because we don't really have kinetic-release branches, and simpler is better)

mikaelarguedas, 2017-05-30:
:+1:
you should be able to change the base branch of https://github.com/ros-planning/navigation/pull/578, otherwise let me know I'll modify it to target lunar

mikaelarguedas, 2017-06-07:
@DLu Anything else I can do to help moving this forward?

mikaelarguedas, 2017-06-28:
@mikeferguson @DLu 2-month anniversary ping. Is there anything preventing this release? (except from the lack of time :wink: )

DLu, 2017-07-13:
https://github.com/ros/rosdistro/pull/15487

https://www.youtube.com/watch?v=U7bGgcHpL1Q

mikaelarguedas, 2017-07-13:
:tada: :tada: thanks @DLu

mikaelarguedas, 2017-07-13:
@DLu I think you need to make a new release or at least a new debinc for the changes from #578 to be applied. Without it the jobs will fail on Debian stretch (http://build.ros.org/view/Queue/job/Lbin_ds_dS64__costmap_2d__debian_stretch_amd64__binary/1/console)

gavanderhoorn, 2017-07-19:
Just ran into this with a ROS Answers poster: [Debian Jessie ROS Install - Dependencies issues](http://answers.ros.org/question/239761) (since migrated to Stretch).

mikeferguson, 2017-07-31:
Once the package v2 stuff is merged, I'll do another release (should be late today or early tomorrow)

closed 2017-07-31 by mikeferguson

mikaelarguedas, 2017-08-05:
@mikeferguson friendly ping

mikeferguson, 2017-08-07:
working on it right now -- release done, just blooming

mikaelarguedas, 2017-08-07:
awesome! thanks @mikeferguson !

mikeferguson, 2017-08-07:
https://github.com/ros/rosdistro/pull/15727