ros-planning/navigation #579: Could NOT find SDL (missing: SDL_LIBRARY SDL_INCLUDE_DIR)

wangzhaodong123, 2017-05-22:
When i make my space,i met this problem:
Could NOT find SDL (missing: SDL_LIBRARY SDL_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:315 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-2.8/Modules/FindSDL.cmake:176 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  navigation/map_server/CMakeLists.txt:11 (find_package)
I'm a freshman,looking for your reply.

AustinDeric, 2017-06-03:
I am having the same issue:

libsdl2-2.0-0 is already the newest version (2.0.4+dfsg1-2ubuntu2).
libsdl2-dev is already the newest version (2.0.4+dfsg1-2ubuntu2).

running ubuntu 16.04 and ROS kinetic

athtest800, 2017-06-10:
Hi! 
`sudo apt-get install libsdl-image1.2-dev` and
`sudo apt-get install libsdl-dev`. These solve the issue.

billdavidzyb, 2017-07-07:
thanks @athtest800 ! That perfectly solved my problem!

boilerbots, 2017-07-11:
I see that map_server won't build and so I changed to SDL2 and had to remove SDL_image because SDL2_image  doesn't install a proper cmake config and it can't be found. Then I had to add explicitly "SDL2_image-2.0" to map_server_image_loader.

mikeferguson, 2017-07-28:
The apt-get rules mentioned above correspond to the rosdeps of the build/run depends on map_server package. So I think this is not a problem with the code/packaging.

For future users coming along, you can install all the required dependencies of your ROS workspace by running the following magic in your workspace (often ~/indigo or ~/kinetic), just be sure to set your rosdistro correctly:
```
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro indigo -y
```

closed 2017-07-28 by mikeferguson

SuTn, 2019-02-20:
> Hi!
> `sudo apt-get install libsdl-image1.2-dev` and
> `sudo apt-get install libsdl-dev`. These solve the issue.

perfect

ibrahiminfinite, 2020-08-12:
> Hi!
> `sudo apt-get install libsdl-image1.2-dev` and
> `sudo apt-get install libsdl-dev`. These solve the issue.

This fixed it, Thanks!

abdouakhad, 2022-08-16:
`sudo apt install cmake libsdl2-dev g++`
This one fix it for me

poeiradonorte, 2023-03-15:
> `sudo apt install cmake libsdl2-dev g++` This one fix it for me

for me too