ros-planning/navigation #583: Local Costmap Using Static Layer gets polluted by fake obstacles

corot, 2017-06-01:
Hi, following #197, I'm currently using a static layer on a rolling window costmap. It works fine if you update the static layer first, then the obstacle one. But for some reasons I need to do the opposite, I want to update the obstacles first. This works with the flag use_maximum = true, but then I get some weird "trailing obstacles" in the first row and column pixels:
[image]
Costmap is fully reset on each iteration, so it seems to me like these "obstacles" are somewhere in memory and brought back with the memory swaps when the robot moves. I can "fix" the problem with this hack:
```
        if (!i || !j)
        {
          master_grid.setCost(i, j, 0);
          continue;
        }
```
on [line 235](https://github.com/ros-planning/navigation/blob/kinetic-devel/costmap_2d/plugins/static_layer.cpp#L325).
Any idea what's going on here?

mikeferguson, 2017-07-28:
This is basically a stream of thoughts, since I should have probably gone to bed a few hours ago:

 * If the static_layer runs updateCosts() after obstacles, doesn't your hack clear out all your obstacles from the master_grid?
 * The LayeredCostmap clears out the master_grid on each iteration (https://github.com/ros-planning/navigation/blob/53087ca5242fb40f06c84a6bcc74057c3470eff7/costmap_2d/src/layered_costmap.cpp#L131).
 * I'm pretty sure rolling static layer always results in updateBounds returning "update the whole cost map"
 * Looking at your picture, the "obstacles" look 1 pixel wide (and then of course inflated). There could be an off-by-one error in static layer updateBounds.

corot, 2017-08-22:
Looks like all was an artifact created by another layer. Disappeared magically after changing it!

closed 2017-08-22 by corot