ros-planning/navigation #584: move_base crashes and get the Inflation and get seen array size is wrong warning

jiangxinyu, 2017-06-02:
I am using the kinetic-devel of the navigation, and some times I will get the warning and move_base crashes. The warning is showing as follows. How can I solve this problem? Please help me.
 [ WARN] [1490717432.681795824]: InflationLayer::updateCosts(): seen_ array size is wrong
[move_base_node-35] process has died [pid 2016, exit code -11, cmd /root/benewake/devel/lib/move_base/move_base /cmd_vel:=/cmd_vel_mux/input/navi __name:=move_base_node __log:=/root/.ros/log/05b6024c-13d1-11e7-ab6b-3adafe3ce91b/move_base_node-35.log].
log file: /root/.ros/log/05b6024c-13d1-11e7-ab6b-3adafe3ce91b/move_base_node-35*.log

fishcu, 2017-06-20:
I am running into the same problem. Very annoying. I am using a very standard navigation setup. The resulting crash is either referenced to as "corrupted double-linked list", or as "invalid free()". Clearly, it is related to some memory trashing bug. Maybe it is related to maps changing the resolution? What would be the right steps to debug this?

fishcu, 2017-06-22:
I think this bug is caused by an oversight in multithreading. I fixed the bug by replacing this function in costmap_2d.cpp:
```
void Costmap2D::resizeMap(unsigned int size_x, unsigned int size_y, double resolution,
                          double origin_x, double origin_y)
{
  size_x_ = size_x;
  size_y_ = size_y;
  resolution_ = resolution;
  origin_x_ = origin_x;
  origin_y_ = origin_y;

  initMaps(size_x, size_y);

  // reset our maps to have no information
  resetMaps();
}
```

With my fixed version:
```
void Costmap2D::resizeMap(unsigned int size_x, unsigned int size_y, double resolution,
                          double origin_x, double origin_y)
{
	boost::unique_lock<mutex_t> lock(*access_);

  size_x_ = size_x;
  size_y_ = size_y;
  resolution_ = resolution;
  origin_x_ = origin_x;
  origin_y_ = origin_y;

  // replace initMaps() to avoid double lock
  delete[] costmap_;
  costmap_ = new unsigned char[size_x * size_y];

  // reset our maps to have no information
  // replace resetMaps() to avoid double lock
  memset(costmap_, default_value_, size_x_ * size_y_ * sizeof(unsigned char));
}
```

The problem was caused because the mutex lock would only get into effect after the size had already been changed. This caused another thread to think the map was already resized, and hence the "seen array size is wrong" error. By puttting a mutex lock at the top of the resize function, the behavior is proper.

mikeferguson, 2018-03-23:
#675 and #686 apply the mutex noted above.

closed 2018-03-23 by mikeferguson