ros-planning/navigation #588: AMCL set particles using pose array

vik748, 2017-07-06:
Hey folks,
I have written a new service for AMCL called 'setParticles' similar to the Global Localization service.  This service accepts a PoseArray msg and initializes the particle filter to this set of poses.  We use this for coming up with schemes to fuse data between multiple robots.  If there are others who would be interested in the feature, I'd be happy to share.

lucbettaieb, 2017-10-25:
If you feel like it could be useful, make a pull request

AchmadFathoni, 2018-07-11:
How is it going?

lucbettaieb, 2018-07-11:
@vik748 ^ <3

vik748, 2018-07-11:
@AchmadFathoni , I have just created a pull request #753 with this feature.
I have to warn you, I had developed this last summer and haven't had a chance to test it with the more recent updates to the code.
Please to test it out and provide your feedback.
Thanks,
Vikrant

closed 2025-06-04 by mikeferguson