ros-planning/navigation #592: AMCL doesn't throw error when it receives maps on the correct topic but not matching the global_frame_id

vik748, 2017-07-20:
Currently,
AMCL seems to proceed as if every thing is good if it receives maps on the specified map_topic even if the frame_id on the map doesn't match global_frame_id.  In this case AMCL essentially is unable to scan map to the map being published and thus the error on the particles keep growing.  A simple fix should be able to warn users that the map being published is not being used by AMCL because of the mismatch.

Cheers,
Vik

mikeferguson, 2017-07-28:
Seems reasonable -- pull request for proposed simple fix would be accepted.

vik748, 2017-07-28:
@mikeferguson I am very new to github.  What is the best way for me to submit the changes?  Do I created a new forked repo and then submit that or do I create a new branch in this repo and request that to be merged?

Thanks,
Vik

mikeferguson, 2017-08-01:
@vik748 new forked repo (you won't have permissions to create branches on this particular repo). Ideally a feature branch within your repo (so you can easily pull in updates to your fork). Then open a PR against this repo. Thanks!

mikeferguson, 2017-08-02:
Fixed in kinetic & lunar

closed 2017-08-02 by mikeferguson