ros-planning/navigation #593: costmap_2d seg faults randomly with footprint_clearing_enabled set to true

akifh, 2017-07-21:
Hi,

I have obstacle layer (`costmap_2d::ObstacleLayer`) with `footprint_clearing_enabled` parameter set to true. Not always, but at a random point in time, I receive segmentation fault, and `move_base` dies. After some debugging and testing with `gdb`, I have this backtrace. Seems like it is related with `setConvexPolygonCost` method. (Ubuntu 16.04 & Kinetic)

```
#0  _int_malloc (av=av@entry=0x7fffc0000020, bytes=bytes@entry=8) at malloc.c:3516
#1  0x00007ffff681f184 in __GI___libc_malloc (bytes=8) at malloc.c:2913
#2  0x00007ffff6e08e78 in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff6536405 in std::vector<costmap_2d::MapLocation, std::allocator<costmap_2d::MapLocation> >::_M_insert_aux(__gnu_cxx::__normal_iterator<costmap_2d::MapLocation*, std::vector<costmap_2d::MapLocation, std::allocator<costmap_2d::MapLocation> > >, costmap_2d::MapLocation const&) ()
   from /opt/ros/kinetic/lib/libcostmap_2d.so
#4  0x00007ffff653336c in costmap_2d::Costmap2D::setConvexPolygonCost(std::vector<geometry_msgs::Point_<std::allocator<void> >, std::allocator<geometry_msgs::Point_<std::allocator<void> > > > const&, unsigned char) () from /opt/ros/kinetic/lib/libcostmap_2d.so
#5  0x00007fffde73ab4c in costmap_2d::ObstacleLayer::updateCosts(costmap_2d::Costmap2D&, int, int, int, int) () from /opt/ros/kinetic/lib/liblayers.so
#6  0x00007ffff654103c in costmap_2d::LayeredCostmap::updateMap(double, double, double) () from /opt/ros/kinetic/lib/libcostmap_2d.so
#7  0x00007ffff65432dc in costmap_2d::Costmap2DROS::updateMap() () from /opt/ros/kinetic/lib/libcostmap_2d.so
#8  0x00007ffff6544316 in costmap_2d::Costmap2DROS::mapUpdateLoop(double) () from /opt/ros/kinetic/lib/libcostmap_2d.so
#9  0x00007ffff50055d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#10 0x00007ffff4dde6ba in start_thread (arg=0x7fffc7c0f700) at pthread_create.c:333
#11 0x00007ffff68a23dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

Any idea on this?

mikeferguson, 2017-07-28:
Without debug symbols, I think this will be fairly impossible to figure out.

akifh, 2017-08-08:
Hopefully, I will try to catch it again with symbols.

akifh, 2017-10-24:
Here it is;

```
Thread 13 "move_base" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc244b700 (LWP 12547)]
0x00007ffff66b2545 in _int_malloc (av=av@entry=0x7fffb4000020,
    bytes=bytes@entry=16) at malloc.c:3727
3727    malloc.c: No such file or directory.

#0  0x00007ffff66b2545 in _int_malloc (av=av@entry=0x7fffb4000020, bytes=bytes@entry=16) at malloc.c:3727
#1  0x00007ffff66b4184 in __GI___libc_malloc (bytes=16) at malloc.c:2913
#2  0x00007ffff6c9de78 in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff4a36bee in __gnu_cxx::new_allocator<costmap_2d::MapLocation>::allocate (this=0x7fffc244a820, __n=2) at /usr/include/c++/5/ext/new_allocator.h:104
#4  0x00007ffff4a369b0 in __gnu_cxx::__alloc_traits<std::allocator<costmap_2d::MapLocation> >::allocate (__a=..., __n=2) at /usr/include/c++/5/ext/alloc_traits.h:182
#5  0x00007ffff4a36682 in std::_Vector_base<costmap_2d::MapLocation, std::allocator<costmap_2d::MapLocation> >::_M_allocate (this=0x7fffc244a820, __n=2) at /usr/include/c++/5/bits/stl_vector.h:170
#6  0x00007ffff4a35a5f in std::vector<costmap_2d::MapLocation, std::allocator<costmap_2d::MapLocation> >::_M_insert_aux (this=0x7fffc244a820, __position=..., __x=...)
    at /usr/include/c++/5/bits/vector.tcc:353
#7  0x00007ffff4a34a26 in std::vector<costmap_2d::MapLocation, std::allocator<costmap_2d::MapLocation> >::push_back (this=0x7fffc244a820, __x=...) at /usr/include/c++/5/bits/stl_vector.h:925
#8  0x00007ffff4a31ece in costmap_2d::Costmap2D::setConvexPolygonCost (this=0xd0c678, polygon=std::vector of length 6, capacity 8 = {...}, cost_value=0 '\000')
    at /home/user1/ros_ws/src/navigation/costmap_2d/src/costmap_2d.cpp:323
#9  0x00007fffe64aafb7 in costmap_2d::ObstacleLayer::updateCosts (this=0xd0c620, master_grid=..., min_i=0, min_j=0, max_i=200, max_j=200)
    at /home/user1/ros_ws/src/navigation/costmap_2d/plugins/obstacle_layer.cpp:434
#10 0x00007ffff4a47481 in costmap_2d::LayeredCostmap::updateMap (this=0xd0bc10, robot_x=-0.056926747687597777, robot_y=0.094615933195685023, robot_yaw=-0.87233441319507743)
    at /home/user1/ros_ws/src/navigation/costmap_2d/src/layered_costmap.cpp:135
#11 0x00007ffff4a4e10e in costmap_2d::Costmap2DROS::updateMap (this=0xd08d40) at /home/user1/ros_ws/src/navigation/costmap_2d/src/costmap_2d_ros.cpp:426
#12 0x00007ffff4a4d941 in costmap_2d::Costmap2DROS::mapUpdateLoop (this=0xd08d40, frequency=10) at /home/user1/ros_ws/src/navigation/costmap_2d/src/costmap_2d_ros.cpp:386
#13 0x00007ffff4a9ae15 in boost::_mfi::mf1<void, costmap_2d::Costmap2DROS, double>::operator() (this=0xd482b8, p=0xd08d40, a1=10) at /usr/include/boost/bind/mem_fn_template.hpp:165
#14 0x00007ffff4a996bd in boost::_bi::list2<boost::_bi::value<costmap_2d::Costmap2DROS*>, boost::_bi::value<double> >::operator()<boost::_mfi::mf1<void, costmap_2d::Costmap2DROS, double>, boost::_bi::list0> (this=0xd482c8, f=..., a=...) at /usr/include/boost/bind/bind.hpp:313
#15 0x00007ffff4a97eb8 in boost::_bi::bind_t<void, boost::_mfi::mf1<void, costmap_2d::Costmap2DROS, double>, boost::_bi::list2<boost::_bi::value<costmap_2d::Costmap2DROS*>, boost::_bi::value<double> > >::operator() (this=0xd482b8) at /usr/include/boost/bind/bind_template.hpp:20
#16 0x00007ffff4a94c08 in boost::detail::thread_data<boost::_bi::bind_t<void, boost::_mfi::mf1<void, costmap_2d::Costmap2DROS, double>, boost::_bi::list2<boost::_bi::value<costmap_2d::Costmap2DROS*>, boost::_bi::value<double> > > >::run (this=0xd48100) at /usr/include/boost/thread/detail/thread.hpp:116
#17 0x00007ffff513d5d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#18 0x00007ffff4f166ba in start_thread (arg=0x7fffc244b700) at pthread_create.c:333
#19 0x00007ffff67373dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

mikeferguson, 2018-03-27:
So, a segmentation fault in malloc probably means heap corruption... that leaves a lot open. I haven't seen many complaints about segfaults in costmap_2d recently, so I'm gonna presume that footprint_clearing_enabled_ really is the deciding thing, that very much limits the code to look at. The only thing that jumps out as a possible culprit would be https://github.com/ros-planning/navigation/blob/53087ca5242fb40f06c84a6bcc74057c3470eff7/costmap_2d/src/costmap_2d.cpp#L335 -- from reading through the code, convexFillCells and getIndex shouldn't cause a problem -- unless the costmap has possibly resized mid-update -- we recently merged a fix for that possibility: https://github.com/ros-planning/navigation/pull/675.

I realize it has been quite a while since this ticket was filed, if you are still fighting this issue, please try that fix out and see if your issues go away?

mikeferguson, 2018-05-14:
Closing due to inactivity after presumed fix. Please reopen if there is  new/additional information.

closed 2018-05-14 by mikeferguson

akifh, 2018-10-11:
@mikeferguson , no problem since #675. Seem like it's fixed. Thanks.