ros-planning/navigation #612: map_server 1.14.1 is failing to export pthread correctly and is breaking downstream packages

tfoote, 2017-08-08:
There are 3 regressions with the same issue: 
http://build.ros.org/job/Kbin_uX64__turtlebot3_navigation__ubuntu_xenial_amd64__binary/
http://build.ros.org/job/Kbin_uX64__stdr_server__ubuntu_xenial_amd64__binary/
http://build.ros.org/job/Kbin_uX64__multi_map_server__ubuntu_xenial_amd64__binary/

[image]

[image]

Example failure:
http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__multi_map_server__ubuntu_xenial_amd64__binary/33/console
```
00:00:52.988 -- Looking for pthread.h
00:00:53.079 -- Looking for pthread.h - found
00:00:53.082 -- Looking for pthread_create
00:00:53.150 -- Looking for pthread_create - not found
00:00:53.151 -- Looking for pthread_create in pthreads
00:00:53.230 -- Looking for pthread_create in pthreads - not found
00:00:53.231 -- Looking for pthread_create in pthread
00:00:53.321 -- Looking for pthread_create in pthread - found
00:00:53.324 -- Found Threads: TRUE  
00:00:53.325 -- Found gtest sources under '/usr/src/gtest': gtests will be built
00:00:53.327 -- Using Python nosetests: /usr/bin/nosetests-2.7
00:00:53.400 -- catkin 0.7.6
00:00:53.575 CMake Error at /opt/ros/kinetic/share/map_server/cmake/map_serverConfig.cmake:141 (message):
00:00:53.575   Project 'multi_map_server' tried to find library '-lpthread'.  The library
00:00:53.575   is neither a target nor built/installed properly.  Did you compile project
00:00:53.575   'map_server'? Did you find_package() it before the subdirectory containing
00:00:53.575   its code is included?
00:00:53.575 Call Stack (most recent call first):
00:00:53.575   /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
00:00:53.575   CMakeLists.txt:4 (find_package)
00:00:53.575 
00:00:53.575 
00:00:53.577 -- Configuring incomplete, errors occurred!
00:00:53.577 See also "/tmp/binarydeb/ros-kinetic-multi-map-server-2.2.2/obj-x86_64-linux-gnu/CMakeFiles/CMakeOutput.log".
00:00:53.577 See also "/tmp/binarydeb/ros-kinetic-multi-map-server-2.2.2/obj-x86_64-linux-gnu/CMakeFiles/CMakeError.log".
00:00:53.581 	cd /tmp/binarydeb/ros-kinetic-multi-map-server-2.2.2
00:00:53.581 	cd obj-x86_64-linux-gnu
00:00:53.581 	"tail -v -n +0 CMakeCache.txt"
```

mikeferguson, 2017-08-10:
@tfoote -- I never got any build failure emails for this, but I am listed as a maintainer. Thoughts?

mikeferguson, 2017-08-10:
Oh, never mind, I see, we broke downstream things

mikeferguson, 2017-08-10:
I'm guessing this affects lunar as well, but none of the "busted" downstream things are released there yet

mikeferguson, 2017-08-10:
@tfoote -- how did you determine it was pthread related?

tfoote, 2017-08-10:
The errors in all the failing builds at the core are: `Project 'multi_map_server' tried to find library '-lpthread'. ` 

Digging a little deeper now and looking inside the generated package config file map_server.pc (from the [testing deb](http://repositories.ros.org/ubuntu/testing/pool/main/r/ros-kinetic-map-server/ros-kinetic-map-server_1.14.1-0xenial-20170807-144208-0800_amd64.deb)) I see the following:

```
prefix=/opt/ros/kinetic

Name: map_server
Description: Description of map_server
Version: 1.14.1
Cflags: -I/opt/ros/kinetic/include -I/usr/include/SDL
Libs: -L/opt/ros/kinetic/lib -lmap_server_image_loader /usr/lib/x86_64-linux-gnu/libSDLmain.a -l:/usr/lib/x86_64-linux-gnu/libSDL.so -l-lpthread -l:/usr/lib/x86_64-linux-gnu/libSDL_image.so
Requires: roscpp nav_msgs tf2
```

Where I've confirmed the pthread export is corrupted.

tfoote, 2017-08-10:
I've confirmed the same issue is in lunar. FYI @mikaelarguedas

The map_serverConfig.cmake generated into the debians has this line that I think is the same issue.
`set(libraries "map_server_image_loader;/usr/lib/x86_64-linux-gnu/libSDLmain.a;/usr/lib/x86_64-linux-gnu/libSDL.so;-lpthread;/usr/lib/x86_64-linux-gnu/libSDL_image.so")`

mikeferguson, 2017-08-11:
@wjwwood -- I think this is related to https://github.com/ros-planning/navigation/pull/560, since that is the only thing adding new libraries. Do you have any idea how we can fix it?

k-okada, 2017-08-11:
This is very interesting and now I understand why our multi_map_server failes to build
`#find_package(SDL REQUIRED) `
https://github.com/ros-planning/navigation/blob/kinetic-devel/map_server/CMakeLists.txt#L12
provides
`SDL_LIBRARY = /usr/lib/x86_64-linux-gnu/libSDLmain.a;/usr/lib/x86_64-linux-gnu/libSDL.so;-lpthread`

This was set by
https://gitlab.kitware.com/cmake/cmake/blob/master/Modules/FindSDL.cmake#L164 as
`    set(SDL_LIBRARY_TEMP ${SDL_LIBRARY_TEMP} ${CMAKE_THREAD_LIBS_INIT})`

and `CMAKE_THREAD_LIBS_INIT` is set to `-lpthread`  by `find_package(Threads)` https://cmake.org/cmake/help/v3.0/module/FindThreads.html

So I am not sure what is the better way to fix that. May be `FindThrads.cmake` set `-lpthread` is ok, but we could not find any issue report on cmake.
https://gitlab.kitware.com/cmake/cmake/issues?scope=all&utf8=%E2%9C%93&state=all&search=FindSDL

One workaround map_server level is to remove `find_package(SDL REQUIRED)`, this might affect some downstream packages. Or more dirty fix is something like
```
list(FIND SDL_LIBRARY "-lpthread" _PTHREAD_INDEX)
if(${_PTHREAD_INDEX} GREATER -1)
  list(INSERT SDL_LIBRARY ${_PTHREAD_INDEX} "pthread")
  MATH(EXPR _PTHREAD_INDEX "${_PTHREAD_INDEX} + 1")
  list(REMOVE_AT SDL_LIBRARY ${_PTHREAD_INDEX})
endif()
```

wjwwood, 2017-08-11:
I came to the same conclusion as @k-okada, I think the offending flag is coming from SDL, and I'm not sure what we're supposed to do about that. I suppose we could try removing the flag as @k-okada suggested.

mikeferguson, 2017-08-14:
Doesn't even look like those downstream packages are building anything against our library -- I'm going to revert the SDL additions for now, so we can get this released.

closed 2017-08-15 by mikeferguson

mikeferguson, 2017-08-15:
New releases:
* https://github.com/ros/rosdistro/pull/15777
* https://github.com/ros/rosdistro/pull/15778

AmeyHande, 2020-03-11:
Hello guys,
I'm also facing same error.
I have latest version of map_server package installed from ros repository for Melodic version.
Any guidance?

commit 0641dc3a (2017-08-15): remove offending library export (fixes #612)

* remove LIBRARY export that is broken in SDL. public builds were
  working fine before this, trying to quickly fix them.
* remove INCLUDE exports as well, since none of our headers actually
  include SDL headers.
commit 24c1e511 (2017-08-15): Merge pull request #614 from ros-planning/fix_k612

remove offending library export (fixes #612)
commit 7d38342d (2017-08-15): remove offending library export (fixes #612)

* remove LIBRARY export that is broken in SDL. public builds were
  working fine before this, trying to quickly fix them.
* remove INCLUDE exports as well, since none of our headers actually
  include SDL headers.
commit a75d24dd (2017-08-15): Merge pull request #615 from ros-planning/fix_l612

remove offending library export (fixes #612)